realsenseai/librealsense #2138: rs-dnn.cpp make error - undefined reference to cv::dnn::experimental_dnn_v1::Net::*

TheMikeyR, 2018-07-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-29-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.14.0                          | 

### Issue Description
I've cloned the master branch from this repository and trying to build it again. 
I'm encountering following error and can't seem to fix the issue, do you have any clue?
```
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `main':
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:28: undefined reference to `cv::dnn::experimental_dnn_v1::readNetFromCaffe(cv::String const&, cv::String const&)'
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:77: undefined reference to `cv::dnn::experimental_dnn_v1::blobFromImage(cv::Mat const&, double, cv::Size_<int> const&, cv::Scalar_<double> const&, bool)'
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:78: undefined reference to `cv::dnn::experimental_dnn_v1::Net::setInput(cv::Mat const&, cv::String const&)'
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:79: undefined reference to `cv::dnn::experimental_dnn_v1::Net::forward(cv::String const&)'
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:28: undefined reference to `cv::dnn::experimental_dnn_v1::Net::~Net()'
/home/mikeyr/dev/librealsense_master/wrappers/opencv/dnn/rs-dnn.cpp:28: undefined reference to `cv::dnn::experimental_dnn_v1::Net::~Net()'
collect2: error: ld returned 1 exit status
wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/build.make:152: recipe for target 'wrappers/opencv/dnn/rs-dnn' failed
make[2]: *** [wrappers/opencv/dnn/rs-dnn] Error 1
CMakeFiles/Makefile2:536: recipe for target 'wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all' failed
make[1]: *** [wrappers/opencv/dnn/CMakeFiles/rs-dnn.dir/all] Error 2
```

I've deleted my build folder, and run 
```
cmake ../ -DBUILD_EXAMPLES=bool:true -DBUILD_PYTHON_BINDINGS=bool:true -DBUILD_CV_EXAMPLES=bool:false -DPYTHON_EXECUTABLE=/home/mikeyr/anaconda3/envs/realsense/bin/python3.6
``` 
```
make -j8
```

dorodnic, 2018-07-25:
Are you working with OpenCV 3.4?

TheMikeyR, 2018-07-25:
@dorodnic yes, 3.4.2 to be exact.

Currently I did a temporarily fix by commenting this line https://github.com/IntelRealSense/librealsense/blob/d45a71da68ad6baa4106f207cf014aa7c7de9df3/wrappers/opencv/CMakeLists.txt#L27
I don't really need the dnn part.

dorodnic, 2018-07-29:
Just in case, I tested latest master with opencv 3.4 branch on Ubuntu 16.04 and everything seem to work. Possibly there is some version collision on your system @TheMikeyR ?

Closing this one since you have a work-around. 
If you choose to revisit the dnn sample and would like more support feel free to open a follow-up issue.

closed 2018-07-29 by dorodnic

amrit-007, 2019-03-15:
Hey @dorodnic ,

I am facing the same issue. Pleae suggest a good solution. Also, I have a custom directory where I am working in, not the github repo.

Here is the output on console,

upsquared@upsquared-UP-APL01:~/amrit/eric/gqcnn/data/realsense/build$ make
Scanning dependencies of target mark
[ 50%] Building CXX object CMakeFiles/mark.dir/src/rs-dnn.cpp.o
[100%] Linking CXX executable bin/mark
CMakeFiles/mark.dir/src/rs-dnn.cpp.o: In function `main':
rs-dnn.cpp:(.text+0x84b): undefined reference to `cv::dnn::dnn4_v20190122::readNetFromCaffe(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
rs-dnn.cpp:(.text+0xda1): undefined reference to `cv::dnn::dnn4_v20190122::blobFromImage(cv::_InputArray const&, double, cv::Size_<int> const&, cv::Scalar_<double> const&, bool, bool, int)'
rs-dnn.cpp:(.text+0xe32): undefined reference to `cv::dnn::dnn4_v20190122::Net::setInput(cv::_InputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, cv::Scalar_<double> const&)'
rs-dnn.cpp:(.text+0xea9): undefined reference to `cv::dnn::dnn4_v20190122::Net::forward(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
rs-dnn.cpp:(.text+0x192a): undefined reference to `cv::dnn::dnn4_v20190122::Net::~Net()'
rs-dnn.cpp:(.text+0x1d81): undefined reference to `cv::dnn::dnn4_v20190122::Net::~Net()'
collect2: error: ld returned 1 exit status
CMakeFiles/mark.dir/build.make:109: recipe for target 'bin/mark' failed
make[2]: *** [bin/mark] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/mark.dir/all' failed
make[1]: *** [CMakeFiles/mark.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Rafipaulino, 2021-02-02:
I've got those errors trying to run rs-dnn.cpp on jetson nano. If we could reopen this topic, would be really helpfull! Log of the error below...

Scanning dependencies of target rs-dnn
[ 50%] Building CXX object CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o
[100%] Linking CXX executable rs-dnn
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `frame_to_mat(rs2::frame const&)':
rs-dnn.cpp:(.text+0x320): undefined reference to `cv::cvtColor(cv::_InputArray const&, cv::_OutputArray const&, int, int)'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `depth_frame_to_meters(rs2::depth_frame const&)':
rs-dnn.cpp:(.text+0x5e4): undefined reference to `cv::Mat::convertTo(cv::_OutputArray const&, int, double, double) const'
rs-dnn.cpp:(.text+0x608): undefined reference to `cv::operator*(cv::Mat const&, double)'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `main':
rs-dnn.cpp:(.text+0x708): undefined reference to `cv::dnn::dnn4_v20200908::readNetFromCaffe(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
rs-dnn.cpp:(.text+0x8dc): undefined reference to `cv::namedWindow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
rs-dnn.cpp:(.text+0x914): undefined reference to `cv::getWindowProperty(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
rs-dnn.cpp:(.text+0xa78): undefined reference to `cv::dnn::dnn4_v20200908::blobFromImage(cv::_InputArray const&, double, cv::Size_<int> const&, cv::Scalar_<double> const&, bool, bool, int)'
rs-dnn.cpp:(.text+0xac8): undefined reference to `cv::dnn::dnn4_v20200908::Net::setInput(cv::_InputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, cv::Scalar_<double> const&)'
rs-dnn.cpp:(.text+0xb10): undefined reference to `cv::dnn::dnn4_v20200908::Net::forward(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
rs-dnn.cpp:(.text+0xd60): undefined reference to `cv::noArray()'
rs-dnn.cpp:(.text+0xd78): undefined reference to `cv::mean(cv::_InputArray const&, cv::_InputArray const&)'
rs-dnn.cpp:(.text+0xe70): undefined reference to `cv::rectangle(cv::_InputOutputArray const&, cv::Rect_<int>, cv::Scalar_<double> const&, int, int, int)'
rs-dnn.cpp:(.text+0xeb0): undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
rs-dnn.cpp:(.text+0xfc8): undefined reference to `cv::rectangle(cv::_InputOutputArray const&, cv::Rect_<int>, cv::Scalar_<double> const&, int, int, int)'
rs-dnn.cpp:(.text+0x103c): undefined reference to `cv::putText(cv::_InputOutputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
rs-dnn.cpp:(.text+0x10a0): undefined reference to `cv::imshow(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&)'
rs-dnn.cpp:(.text+0x10c0): undefined reference to `cv::waitKey(int)'
rs-dnn.cpp:(.text+0x1234): undefined reference to `cv::dnn::dnn4_v20200908::Net::~Net()'
rs-dnn.cpp:(.text+0x1594): undefined reference to `cv::dnn::dnn4_v20200908::Net::~Net()'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::Mat(int, int, int, void*, unsigned long)':
rs-dnn.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x150): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
rs-dnn.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x244): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
rs-dnn.cpp:(.text._ZN2cv3MatC2EiiiPvm[_ZN2cv3MatC5EiiiPvm]+0x2cc): undefined reference to `cv::Mat::updateContinuityFlag()'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::Mat(cv::Size_<int>, int, void*, unsigned long)':
rs-dnn.cpp:(.text._ZN2cv3MatC2ENS_5Size_IiEEiPvm[_ZN2cv3MatC5ENS_5Size_IiEEiPvm]+0x154): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
rs-dnn.cpp:(.text._ZN2cv3MatC2ENS_5Size_IiEEiPvm[_ZN2cv3MatC5ENS_5Size_IiEEiPvm]+0x248): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
rs-dnn.cpp:(.text._ZN2cv3MatC2ENS_5Size_IiEEiPvm[_ZN2cv3MatC5ENS_5Size_IiEEiPvm]+0x2d0): undefined reference to `cv::Mat::updateContinuityFlag()'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::~Mat()':
rs-dnn.cpp:(.text._ZN2cv3MatD2Ev[_ZN2cv3MatD5Ev]+0x34): undefined reference to `cv::fastFree(void*)'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::release()':
rs-dnn.cpp:(.text._ZN2cv3Mat7releaseEv[_ZN2cv3Mat7releaseEv]+0x58): undefined reference to `cv::Mat::deallocate()'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::operator()(cv::Rect_<int> const&) const':
rs-dnn.cpp:(.text._ZNK2cv3MatclERKNS_5Rect_IiEE[_ZNK2cv3MatclERKNS_5Rect_IiEE]+0x24): undefined reference to `cv::Mat::Mat(cv::Mat const&, cv::Rect_<int> const&)'
CMakeFiles/rs-dnn.dir/rs-dnn.cpp.o: In function `cv::Mat::operator=(cv::Mat&&)':
rs-dnn.cpp:(.text._ZN2cv3MataSEOS0_[_ZN2cv3MataSEOS0_]+0xf4): undefined reference to `cv::fastFree(void*)'
collect2: error: ld returned 1 exit status
CMakeFiles/rs-dnn.dir/build.make:94: recipe for target 'rs-dnn' failed
make[2]: *** [rs-dnn] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rs-dnn.dir/all' failed
make[1]: *** [CMakeFiles/rs-dnn.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

andzhouwei, 2023-07-18:
help!!!
opencv3.42
ubuntu18.4

rs-dnn.cpp:(.text._ZN3rs216processing_block5startINS_11frame_queueEEEvT_[_ZN3rs216processing_block5startINS_11frame_queueEEEvT_]+0x76)：对‘rs2_start_processing’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/rs-dnn.dir/build.make:94: recipe for target 'rs-dnn' failed
make[2]: *** [rs-dnn] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rs-dnn.dir/all' failed
make[1]: *** [CMakeFiles/rs-dnn.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2