realsenseai/librealsense #2139: Wrong RS2_FORMAT_DISPARITY16 description

pavloblindnology, 2018-07-25:
This line should be `The depth in meters is equal to depth 1 / (scale * pixel value).`
https://github.com/IntelRealSense/librealsense/blob/d45a71da68ad6baa4106f207cf014aa7c7de9df3/include/librealsense2/h/rs_sensor.h#L59

RealSense-Customer-Engineering, 2018-07-26:
[Realsense Customer Engineering Team Comment]
@pavloblindnology
The description is wrong and I've created pull request, the right description should be "(baseline * focal length) / (scale * pixel value)".

https://github.com/IntelRealSense/librealsense/pull/2152

pavloblindnology, 2018-07-26:
For DISPARITY16 I've got the next correspondence after testing in 720hd mode:
RS2_FORMAT_DISPARITY16 value: ~3000
RS2_FORMAT_Z16 value: ~330 (0.33m)
My formula gives depth(m) = 1 / (scale * pixel value) = 1 / (0.001 * 3000) = 0.33m which is perfectly well.
Your formula gives depth(m) = (baseline * focal length) / (scale * pixel value) = 0.05 * 640 / (0.001 * 3000) = 10.66m which is wrong.
Please check.

ev-mp, 2018-07-26:
Hello @pavloblindnology ,
Librealsense use of the disparity term is consistent with the standard definition for stereo vision:  [Disparity(displacement)=focal_length*stereo_baseline/Z](https://en.wikipedia.org/wiki/Computer_stereo_vision)

Note that DISPARITY_16 is a legacy format and is not natively supported in librealsense2 , i.e. you cannot request and receive frames of this type.
The current SDK v2 does use a related DISPARITY_32 format internally, though the usage is limited to post-processing blocks to obtain uniform data filtering.

ev-mp, 2018-07-26:
Hello @pavloblindnology , 
The values you obtain by modifying this control are not disparity but depth (Z16) values with a different multiplier. 
There is a simple test - disparity is intrinsically in inverse proportion to depth, i.e as depth values increase disparity decreases and vice versa.
Using realsense-viewer it is possible to check that with `disparityMode=1` this is not the case. (values change proportionally and not inverse-proportionally)

Can you share a link to the documentation you used ? - it is possible that it was misleading and we need to update it. 
Thanks you

pavloblindnology, 2018-07-27:
And this simple test (which I, as a computer vision engineer, did from the
very beginning) shows that it's in inverse proportion.
Please see attached screenshots - two sets for near and far pixels.
Set 1: pixel [710, 87], raw disparity value - 392, raw depth value - 2543,
Set 2: pixel [570, 684], raw disparity value - 3003, raw depth value - 340,
Depth sensor parameters: D435, 1280x720, fx = 638.0615844726562, baseline
= 0.04993824660778046.
You can easily check which relationships between them are correct.

And your documentation for RS2_FORMAT_DISPARITY16 is really incorrect, so I
had to figure out the correct relationship.

Pavlo

pavloblindnology, 2018-08-02:
Any update on this?

pavloblindnology, 2018-08-02:
The formal definition is not applicable to RS2_FORMAT_DISPARITY16 because uint16 has limited values range.
The formal definition disparity_value = (baseline * focal length) / (scale * depth) 
gives us disparity_value = 0.05 * 640 / (0.001 * 0.2m) = 160000 which is way out of uint16 scope.

ev-mp, 2018-08-09:
Hello @pavloblindnology ,
I can update that with the current firmware versions when applied, the produced disparity is transformed as follows:
`depth = (baseline * focal_length *32) / disparity`  (all units shall agree)

As mentioned above,  RS2_FORMAT_DISPARITY16 was supported in the legacy SDK, but not librealsense2. we'll update its documentation to prevent further confusion.

We'll further discuss it internally to see whether it shall be added to supported features. 
I the meanwhile it remains available as a developer option, and as such might be subject for changes in future versions.

pavloblindnology, 2018-08-09:
Hello @ev-mp Thanks for clarification. Want to note that RS2_FORMAT_DISPARITY16 can be useful to minimize transportation and saving to disk cost in ROS infrastructure.

ev-mp, 2018-08-09:
As with regards to the advanced controls available, there is an additional resource that elaborates on some of the `advanced mode` controls that should be utilized for fine-tuning:
https://realsense.intel.com/wp-content/uploads/sites/63/BKM-For-Tuning-D435-and-D415-Cameras-Webinar_Rev3.pdf  that imo, is more application-oriented.

agrunnet, 2018-08-09:
Just to add to that comment. Please note the factor 32 in the equation that was missing from your original equation. Basically when the output is in disparity it is in units of 1/32 pixels.

closed 2018-08-24 by RealSense-Customer-Engineering