realsenseai/librealsense #2141: D435 FOV question

omerbrandis, 2018-07-25:
hello,

i'm running rs v2.14 , and firmware 5.9.11 on a D435.

according to "[D400 Series Product Family rev003 from 07/2018](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf)"
page 50,

**Color Camera FOV H:69.4±3 ,V:42.5±2** 

but i get different results using 
`
rs2::config   myconfig;

 MyPipelineProfile = myp.start(myconfig);

auto ColorStream = MyPipelineProfile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();

ColorIntrinsics = ColorStream.get_intrinsics();

rs2_fov(&ColorIntrinsics, fov);`

**i get values 
horizonal fov = 54.732
vertical fov = 42.4115**

Can someone shed some light on this ?

Thanks,
Omer.

MartyG-RealSense, 2018-07-25:
the documentation on the Projection system for the 400 Series cameras states that "image FOV can vary".

The forum link below discussed this subject recently when someone else queried the FOV values they were getting on their D435:

https://communities.intel.com/message/555626#555626

omerbrandis, 2018-07-25:
Hello,
I've reviewed the links provided, but i didn't see an explicit answer for my problem in them.

( the doc's claim a possible variation of +/- of 3 degrees , but i got a 15 deg variation on my horizontal fov ...)

Thanks,
Omer.

MartyG-RealSense, 2018-07-25:
I do not have any further data about the FOV to provide.  Hopefully one of the Intel support members on this GitHub can help you further .  My apologies - thanks for your patience!

RealSense-Customer-Engineering, 2018-07-27:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

let me check on this.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

would like to know which resolution when you read the FOV value below?
thanks.

omerbrandis, 2018-08-12:
Hello 

640X480.

omer.

RealSense-Customer-Engineering, 2018-08-15:
[Realsense Customer Engineering Team Comment]
hi @omerbrandis,

per checked, the FOV in the datasheet is for HD resolution. So this is expected with the resolution you are using.

closed 2018-08-16 by omerbrandis