realsenseai/librealsense #2143: realsense_app A/libc: Fatal signal 11 (SIGSEGV), code 2, fault addr 0xa03e4000 in tid 10748 (Thread-9464)

fotsing365, 2018-07-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version | Android 6.0.1| 

| Platform                                 |   |
| SDK Version                            | 2.12.0                        | 
| Language                            |  C++                        | 
| Segment			|  Rooted Lenovo YT3-850F                        | 

After following intructions given in #2075 I successful fix the issue I described in #2008. But after launching realsense_app it displays the 3D map for few seconds and crashes. The logcat look like this 
07-25 09:08:00.949 2527-2527/? I/art: Late-enabling -Xcheck:jni
07-25 09:08:00.965 2527-2527/? I/Zygote: ZygoteInit::mMethod=public static void android.app.ActivityThread.main(java.lang.String[])
07-25 09:08:01.090 2527-2527/? W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_dependencies_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_dependencies_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.335 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_0_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_0_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.417 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_1_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_1_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.492 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_2_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_2_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.558 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_3_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_3_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.624 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_4_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_4_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.703 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_5_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_5_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.784 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_6_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_6_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.853 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_7_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_7_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.920 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_8_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_8_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:01.992 2527-2527/com.example.realsense_app W/art: Failed execv(/system/bin/dex2oat --runtime-arg -classpath --runtime-arg  --instruction-set=arm --instruction-set-features=smp,div,-atomic_ldrd_strd --runtime-arg -Xrelocate --boot-image=/system/framework/boot.art --runtime-arg -Xms64m --runtime-arg -Xmx512m --instruction-set-variant=cortex-a7 --instruction-set-features=default --dex-file=/data/app/com.example.realsense_app-2/split_lib_slice_9_apk.apk --oat-file=/data/dalvik-cache/arm/data@app@com.example.realsense_app-2@split_lib_slice_9_apk.apk@classes.dex) because non-0 exit status
07-25 09:08:02.001 2527-2527/com.example.realsense_app I/InstantRun: starting instant run server: is main process
07-25 09:08:02.387 2527-2527/com.example.realsense_app W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
07-25 09:08:02.510 2527-2527/com.example.realsense_app I/art: Rejecting re-init on previously-failed class java.lang.Class<android.support.v4.view.ViewCompat$OnUnhandledKeyEventListenerWrapper>
[... cut ...]
07-25 09:08:02.828 2527-2612/com.example.realsense_app I/OpenGLRenderer: Initialized EGL, version 1.4
**07-25 09:08:02.995 2527-2616/com.example.realsense_app A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0xb895e000 in tid 2616 (Thread-9953)**
07-25 09:08:03.043 2527-2527/com.example.realsense_app W/art: Before Android 4.1, method int android.support.v7.widget.DropDownListView.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
 
[image]

fotsing365, 2018-07-26:
According my investigation, using **__android_log_print** messages I have noticed that the problem is in the native code.  It occurs during the instruction `profile = pipe.start(config);`. Indeed, it's seem like after this instruction there is a context which is initialazed and which dies just few seconds later before  the instruction  pipe.stop(); which follows an potetial infinite loop `WHILE`. 

**CASES:**
case 1-:  If comment `While` loop including its content the app works fine
           `/* while (streaming) {
                     rs2::frameset data = pipe.wait_for_frames();
                     auto rgb_vid_frame = colorizer.colorize(data);
 
                     env->SetByteArrayRegion(arr, 0, rgb_frame_size,
                                             reinterpret_cast<const jbyte *>(rgb_vid_frame.get_data()));
 
                     env->CallStaticVoidMethod(handlerClass, onFrame, arr, width,
                                               height);
                     
                 }*/`
Case2: If I comment only the `WHILE` loop contains, the apps will crash by returning same error like before
               ` while (streaming) {
                   /*  rs2::frameset data = pipe.wait_for_frames();
                     auto rgb_vid_frame = colorizer.colorize(data);
 
                     env->SetByteArrayRegion(arr, 0, rgb_frame_size,
                                             reinterpret_cast<const jbyte *>(rgb_vid_frame.get_data()));
 
                     env->CallStaticVoidMethod(handlerClass, onFrame, arr, width,
                                               height);
                  */   
                 }`

jamescartercamer, 2018-07-30:
Hi @fotsing365.
have you advanced on this issue ? I face exactly the same problem

jamescartercamer, 2018-07-31:
Hi @jamescartercamer not yet. But like I said the issue is from  this instruction `profile = pipe.start(config);` in the native-lib.cpp

andophine, 2018-09-25:
Suggest replace the sample code that start the pipe with below line
    
    rs2::pipeline pipe;
    rs2::config cfg;
    rs2::colorizer colorizer;
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    cfg.enable_stream(RS2_STREAM_INFRARED, 640, 480, RS2_FORMAT_RGB8, 30);
    bool test = cfg.can_resolve(pipe);
    if (test == false){
        cfg.disable_all_streams();
        cfg.enable_stream(RS2_STREAM_DEPTH, 424, 240, RS2_FORMAT_Z16, 30);
        cfg.enable_stream(RS2_STREAM_INFRARED, 424, 240, RS2_FORMAT_Y8, 30);
        
    }
    rs2::pipeline_profile profile = pipe.start(cfg);

to avoid device that not support USB3.0. 
However, even you can make streaming run, the SDK is still very unstable on Android. You can try to integrate with PCL library, everything won't work. Without the RealSense SDK, PCL Android work correctly.

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
pls reference to the new Android example:

https://github.com/IntelRealSense/librealsense/pull/2489

RealSense-Customer-Engineering, 2018-10-10:
[Realsense Customer Engineering Team Comment]
before the official new Android example(https://github.com/IntelRealSense/librealsense/pull/2489) available, pls reference to the following Android example to

troubleshooting the issue on Android:

https://github.com/zhouwg/irsa/tree/master/examples/rs_multicam

RealSense-Customer-Engineering, 2018-10-15:
[Realsense Customer Engineering Team Comment]
@fotsing365，

 can we close the ticket accordingly? Please update. Thanks!

RealSense-Customer-Engineering, 2018-10-15:
[Realsense Customer Engineering Team Comment]
Hi @fotsing365,

pls reference to:

https://github.com/zhouwg/irsa/blob/master/docs/IRSA_introduction.txt

https://github.com/zhouwg/irsa/blob/master/how-to-build-irsa.txt

https://github.com/zhouwg/irsa/blob/master/examples/how-to-build-examples.txt

https://github.com/zhouwg/irsa/blob/master/examples/rs_multicam/app/src/main/cpp/native-lib.cpp#L457

 can we close the ticket accordingly? Please update. Thanks too much!

RealSense-Customer-Engineering, 2018-10-30:
[Realsense Customer Engineering Team Comment]
@fotsing364 Is the issue resolved? Looking forward to your update. Thanks!