realsenseai/librealsense #2149: Best way to save realsense data

uddhavbhattarai, 2018-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2018-07-25:
If you tried to save all image and ply data in realtime, the performance of the stream would likely be very poor due to the processing load.  It may be better to store the entire captured data in memory and then perform saving of the frames and any postprocessing at the end.

 The Librealsense team created a function called Keep() for this kind of memory storage.

https://github.com/IntelRealSense/librealsense/issues/1000

dorodnic, 2018-07-26:
Writing to PLY is very slow. 
Best way to save realsense data without losing information would be the ROS-bag format.
Our built-in recorder will collect everything you need, and it is fairly well optimized, at least when running in Release. 
You can then extract individual frames / models as a post-processing step.

uddhavbhattarai, 2018-07-27:
Thank you for the inputs.

closed 2018-07-27 by uddhavbhattarai