realsenseai/librealsense #2150: Project needs to be in a directory for snap packaging

SteveMacenski, 2018-07-25:
The librealsense project needs to be in a subdirectory in the repo in order to include and build it as a snap package. It requires catkin projects to be in a subdirectory (which is also a ROS REP I believe) as you can't build the root source space. 

Can librealsense be moved into an appropriately named subdirectory to aid in this?

SteveMacenski, 2018-07-26:
Hi,

So the issues is just that the contents of librealsense needs to be in a directory. When you git clone into source, it assumes the repo will be a meta catkin package or contain a catkin package within it. By ROS standards, that's how all repositories should be built, but its a 50/50 if people actually do that. For instance, realsense ROS repository does this correctly and doesnt have any issues (inside the repo, there's a directory that is the project).

So snap packaging has a catkin plugin which assumes that people followed the catkin design guidelines in making projects inside of repositories and not that the repository itself is a project. As a result, when you clone in, the contents of the repo are dumped into source and then there's no directory surrounding it to catch the package.xml file to index it as its the root of the workspace.

The issue isn't as much about snap packaging as it is about catkin. The issue arises because snapcraft assumes compliance.

SteveMacenski, 2018-07-26:
```  workspace_folder/
    src/
      my application folder  <-- git repo
        package_a/
        package_b/
```

This is an example of how its supposed to be layed out in a answer from Dirk from OSRF. 

Here is the doc for the catkin plugin in snapcraft, though the beef here is more with catkin than snapcraft itself

https://docs.snapcraft.io/reference/plugins/catkin

RealSense-Customer-Engineering, 2018-12-17:
[Realsense Customer Engineering Team Comment]
Per internal discussion, no plan to change since the effort is too big to justify changing the directory structure of the SDK library. It is used by many users and the change might be disruptive.

closed 2018-12-17 by RealSense-Customer-Engineering

SteveMacenski, 2018-12-17:
Its not to directory structure, its just putting it into a directory instead in the of the repo