realsenseai/librealsense #2155: Sensor Control Sample Throwing Exceptions

octave1, 2018-07-26:
I just recently updated my librealsense2 SDK version from 2.10.4 to 2.13.0. When I went to run my previously built code to check for compatibility I noticed that any time I tried to use set_option() it would throw an exception, as shown below. 

[image]

[image]

(Note: I am using the d415.)

I dug a bit deeper and noticed that the function  "device.query_sensors(); " which had previously returned two elements (one for rgb sensor and one for ir sensor) was only returning one element. 

I then decided to run the sensor control sample code found here: https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/rs-sensor-control.cpp

and found that after selecting any device, the following error was thrown and the program terminated (Previously I could view valid sensor options).

[image]

The only change I have made was updating the SDK from v 2.10.4 -> v 2.13.0. Are the methods used to acquire devices and sensors previously implemented in older SDK versions no longer supported? Has anyone encountered similar issues?

dorodnic, 2018-07-29:
Hi @octave1 
Can you see the error message associated with the `wrong_api_call_sequence_error` exception? 
Are you building the SDK from source or using the pre-built DLL & executables? 
Are you getting this error only in a specific application or in all applications provided under [releases](https://github.com/IntelRealSense/librealsense/releases)?

octave1, 2018-07-29:
Hi @dorodnic ,
I recently read through some forum posts, which led me to believe that the pre-built binaries weren't being updated alongside the source code. I downloaded the latest SDK release and built it from source. Now everything seems to be working fine. 
This issue can now be closed. 
Thanks,
octave1

closed 2018-07-29 by octave1