realsenseai/librealsense #2161: C++ hardware reset (not ROS)

Vivikyung, 2018-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (V2.13) | 
| Operating System & Version |   { Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  16.04.1                                      | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.14.1 }                          | 
| Language                            |  {C++ with OpenCV }                          | 
| Segment			|                          | 

### Issue Description
I'm having a similar issue to some others where when I plug my camera in once, it works fine, but if I stop the stream and restart it I get a "No Frames Received error" (on the Realsense-Viewer. It just hangs on all the examples / my own script). The only workaround I've figured is to unplug and replug the camera back in, and then it works just fine again... once. It consistently only works once.

I know how to manual reset it using the Realsense-Viewer, but I can't figure out how to code wise. (My workaround currently is to just hardware reset the cameras at the beginning of my script.)

If I try:
> rs2::device dev = device_you_are_using();
> dev.hardware_reset(); //Will disconnect the camera and reconnect it.

I get a "'device_you_are_using' was not declared in scope" error and it fails to compile.

Similarly, if I try: 
> rs2::context ctx;
> rs2::device dev = get_device_from_context(ctx);
> dev.hardware_reset();
> rs2::device_hub hub(ctx);
> dev = hub.wait_for_device(); //Note that device hub will get any device, if you have more than 1 connected it could return the other device

I get a "'get_device_from_context' was not declared in scope" error and it also fails.

Could someone please explain how I can implement this in C++? There is very little / no documentation on hardware_reset(). Thank you!

RealSense-Customer-Engineering, 2018-07-30:
[Realsense Customer Engineering Team Comment]
hi @Vivikyung,

I think i didn't see realsense-viewer on Ubuntu have this issue, would like to reproduce this issue first, so please tell which camera you are using, fw version, so I can have a try, thanks.

dorodnic, 2018-07-30:
Hi @Vivikyung 
You can do something like:
```cpp
rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();
```
This will only reset the **first** device is you have several connected.

Vivikyung, 2018-07-31:
@dorodnic Thank you for your reply!
@RealSense-Customer-Engineering  I had both a 415 and 435 but I think I just had a faulty hub because once I switched it is now working fine.

closed 2018-07-31 by Vivikyung

BenBergman, 2019-09-19:
I was experiencing a similar issue. I got errors when trying to get the hub and wait for device, but just removing those last two lines from dorodnic's snippet worked for me.

themancalledjakob, 2020-02-18:
> Hi @Vivikyung
> You can do something like:
> 
> ```c++
> rs2::context ctx;
> rs2::device dev = ctx.query_devices().front(); // Reset the first device
> dev.hardware_reset();
> rs2::device_hub hub(ctx);
> dev = hub.wait_for_device();
> ```
> 
> This will only reset the **first** device is you have several connected.

Sometimes this hardware_reset fails for me to resolve the issue, and even rebooting does not reliably help to bring the cameras back.
Usually I use the realsense cameras in an installation setting, where it is impossible to physically unplug and replug the USB cables. For this I collected some methods to script a usb reset and so far this **always** worked for me on Ubuntu: https://github.com/themancalledjakob/usbreset

Is there any reason this is not a good idea, or why a similar functionality is not implemented in the realsense sdk?

FredM67, 2022-02-04:
> Hi @Vivikyung You can do something like:
> 
> ```c++
> rs2::context ctx;
> rs2::device dev = ctx.query_devices().front(); // Reset the first device
> dev.hardware_reset();
> rs2::device_hub hub(ctx);
> dev = hub.wait_for_device();
> ```
> 
> This will only reset the **first** device is you have several connected.

Well, this could work .... IF I get a device !!
In my case, the list is empty.
I also tried to "resetall" script, does not work either ! I'm stucked...