realsenseai/librealsense #2169: Timestamps and Frame Counter

rafaor29, 2018-07-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.13.0-45-generic                                         | 
| SDK Version | 2.14.0 |
| Platform                                 | PC |

### Issue Description

Hi there, i was wondering how can i set the timestamp and the frame counter for the captures made by 2 d415 cameras so i can compare them and check if the HW sync has been done properly.
Right now i'm following the save-to-disk example inside of the following loop:

 `for (auto&& dev : ctx.query_devices()) // Query the list of connected RealSense devices`

but it doesn't work properly, it only saves the image captured by the first camera.

Thanks for the help.

dorodnic, 2018-08-05:
Hi @rafaor29 
I made some changes to `rs-save-to-disk` for make it save snapshots from all connected devices concurrently:
```cpp
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API

#include <fstream>              // File IO
#include <iostream>             // Terminal IO
#include <sstream>              // Stringstreams
#include <thread>               // For thread

// 3rd party header for writing png files
#define STB_IMAGE_WRITE_IMPLEMENTATION
#include "stb_image_write.h"

// Helper function for writing metadata to disk as a csv file
void metadata_to_csv(const rs2::frame& frm, const std::string& filename);

void save_from_camera(std::string serial)
{
    // Declare depth colorizer for pretty visualization of depth data
    rs2::colorizer color_map;

    // Declare RealSense pipeline, encapsulating the actual device and sensors
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device(serial);
    // Start streaming with default recommended configuration
    pipe.start(cfg);

    // Capture 30 frames to give autoexposure, etc. a chance to settle
    for (auto i = 0; i < 30; ++i) pipe.wait_for_frames();

    // Wait for the next set of frames from the camera. Now that autoexposure, etc.
    // has settled, we will write these to disk
    for (auto&& frame : pipe.wait_for_frames())
    {
        // We can only save video frames as pngs, so we skip the rest
        if (auto vf = frame.as<rs2::video_frame>())
        {
            auto stream = frame.get_profile().stream_type();
            // Use the colorizer to get an rgb image for the depth stream
            if (vf.is<rs2::depth_frame>()) vf = color_map(frame);

            // Write images to disk
            std::stringstream png_file;
            png_file << "rs-save-to-disk-output-" << serial << "-" << vf.get_profile().stream_name() << ".png";
            stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
                vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
            std::cout << "Saved " << png_file.str() << std::endl;

            // Record per-frame metadata for UVC streams
            std::stringstream csv_file;
            csv_file << "rs-save-to-disk-output-" << serial << "-" << vf.get_profile().stream_name()
                << "-metadata.csv";
            metadata_to_csv(vf, csv_file.str());
        }
    }
}

// This sample captures 30 frames and writes the last frame to disk.
// It can be useful for debugging an embedded system with no display.
int main(int argc, char * argv[]) try
{
    std::vector<std::thread> threads;
    for (auto dev : rs2::context().query_devices())
    {
        auto serial = dev.query_sensors().front().get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
        threads.emplace_back([serial]() {
            save_from_camera(serial);
        });
    }
    for (auto&& t : threads) t.join();

    return EXIT_SUCCESS;
}
catch(const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch(const std::exception & e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

void metadata_to_csv(const rs2::frame& frm, const std::string& filename)
{
    std::ofstream csv;

    csv.open(filename);

    //    std::cout << "Writing metadata to " << filename << endl;
    csv << "Stream," << rs2_stream_to_string(frm.get_profile().stream_type()) << "\nMetadata Attribute,Value\n";

    // Record all the available metadata attributes
    for (size_t i = 0; i < RS2_FRAME_METADATA_COUNT; i++)
    {
        if (frm.supports_frame_metadata((rs2_frame_metadata_value)i))
        {
            csv << rs2_frame_metadata_to_string((rs2_frame_metadata_value)i) << ","
                << frm.get_frame_metadata((rs2_frame_metadata_value)i) << "\n";
        }
    }

    csv.close();
}
```
The implication of hardware-sync is on the visual content of the frames, and to lesser extent on timestamp / frame-number (each camera has its own clock)

rafaor29, 2018-08-06:
@dorodnic thank you very much, this is very clarifying.

closed 2018-08-06 by rafaor29