realsenseai/librealsense #2170: Couldn't find any package by glob 'intel-realsense-dfu*'

joelmaranhao, 2018-07-30:
Will be using RealSense D435 and I am following [these instructions](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf) to upgrade the firmware.

Any pointers?

# My Ubuntu 16
```bash
~# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.5 LTS
Release:	16.04
Codename:	xenial
```

# Update

```bash
~# sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
Executing: /tmp/tmp.eQ0W3pqSHw/gpg.1.sh --keyserver
keys.gnupg.net
--recv-key
6F3EFCDE
gpg: requesting key 6F3EFCDE from hkp server keys.gnupg.net
gpg: key 6F3EFCDE: ""CN = Intel(R) Intel(R) Realsense", O=Intel Corporation" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
```

## Failed to fetch http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages  404  Not Found

```bash
~# sudo apt-get update
Hit:1 http://security.ubuntu.com/ubuntu xenial-security InRelease
Ign:2 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial InRelease                                                                     
Hit:3 http://us.archive.ubuntu.com/ubuntu xenial InRelease                                                                                            
Ign:4 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial Release
Get:5 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]
Ign:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Get:9 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons
Ign:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons
Ign:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons
Ign:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons
Ign:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons
Err:6 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 Packages
  404  Not Found
Ign:7 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main i386 Packages
Ign:8 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main all Packages
Ign:10 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en_US
Ign:11 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main Translation-en
Ign:12 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main amd64 DEP-11 Metadata
Ign:13 http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial/main DEP-11 64x64 Icons                                                                                                                                 
Fetched 216 kB in 6s (34.1 kB/s)                                                                                                                                                                                                 
Reading package lists... Done
W: The repository 'http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

## Couldn't find any package by glob 'intel-realsense-dfu*'

```bash
~# sudo apt-get install intel-realsense-dfu* 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package intel-realsense-dfu*
E: Couldn't find any package by glob 'intel-realsense-dfu*'
E: Couldn't find any package by regex 'intel-realsense-dfu*'

```

RealSense-Customer-Engineering, 2018-07-31:
[Realsense Customer Engineering Team Comment]
hi @joelmaranhao,

just verified the steps to install dfu tool and it's working good, so no idea about your issue.
can you please provide your /etc/apt/sourde.list.d/realsense*.list for me to take a look?
after test, i think you have typo in the list, it should be "realsense-hw-public" not "realsense-hwpublic", please have a try

joelmaranhao, 2018-08-01:
Here it is

```bash
root@qb57:~# vim /etc/apt/sources.list.d/realsense-public.list 
deb http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial main
```

Fixing the typo RESOLVED

closed 2018-08-01 by joelmaranhao

hello-sea, 2019-03-29:
Thank you very much, I solved the same problem.

Please follow the process below to try(note the spelling): 
```shell
# (1)
echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
# (2)
sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
# (3)
sudo apt-get update
# (4)
sudo apt-get install intel-realsense-dfu
```