realsenseai/librealsense #2172: pyrealsense2 enable_record_to_file() help

upxela, 2018-07-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.14.0                         | 
| Language                            |  Python                        | 

### Issue Description
My ultimate goal is to be able to preview a video stream while recording as well. While using `enable_record_to_file()`, I realized I was dropping frames.

I added a `keep()` call which allowed me to not drop frames, but now when I try to record a video of longer length (let's say 20 seconds), the output streams (in the rosbag file) are only around 9 seconds long.

My current code is the following:

`pipeline = rs.pipeline()` 
`config = rs.config()`  
`config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)`  
`config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)`  
`config.enable_record_to_file('test.bag')`
`pipeline.start(config)`  
`try:`  
    `start = time.time()  `
    `while time.time()-start < 20:`  
         `pipeline.wait_for_frames().keep()`  
`finally:  `
    `pipeline.stop()  `

Is there a way to avoid this? Or is there a better way to write to a file in general?

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
@upxela
To preview while streaming, you don't need to use keep, you can just acquire the depth from pipeline and simply display it in your window, the record to file will handle the work to save image to ros bag file.  You can refer to the C++ example rs-record-playback.cpp.

upxela, 2018-08-09:
@RealSense-Customer-Engineering 
Thanks so much for getting back to me.

The problem right now is that the enable_record_to_file() function drops frames unless I use the keep() function. However, using the keep() function often clips the final saved ros bag short significantly.

I'm using python3.7 on Windows 10.

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
@upxela
Can you provide the py file for me to reduplicate your issue?

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
Any update?

upxela, 2018-08-24:
@RealSense-Customer-Engineering 

Thanks for getting back to me. I was originally using Python 3.7, building pyrealsense2 from source.
I've realized that if I used Python 2.7 and use the PyPI distribution of pyrealsense2, my problem is gone. The script that I can see a significant decrease in captured frames is attached.

If you run the script, you can visibly notice that the program slows down.

[python3problem.zip](https://github.com/IntelRealSense/librealsense/files/2319468/python3problem.zip)

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
@upxela
I've tried your code with my python 3.6.3, I didn't see any slow down as you mentioned, please refer to my two captured images. The play back in realsense viewer is 34 seconds and from my console I also runt it with around 34 second. I didn't see the problem.

carypai, 2018-08-28:
[image]
[image]

upxela, 2018-08-29:
Might be a problem with the way I built it then! I'll take a look and see if I get better results

RealSense-Customer-Engineering, 2018-09-20:
[Realsense Customer Engineering Team Comment]
Hi @upxela,
is this still an issue with your setup?
best

closed 2018-12-12 by RealSense-Customer-Engineering