realsenseai/librealsense #2173: C depth example "unsupported descriptor subtype: 3" error

jlbeebout, 2018-07-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | {05.09.14.00} | 
| Operating System & Version |   {Mac OSX High Sierra 10.13.6}  | 
| Platform                                 | Mac  |
| SDK Version                            |  { legacy / 2.14.1 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {Other }                          | 

### Issue Description
I am having trouble getting the Realsense camera to work with the C depth and C distance examples. The camera works with other code examples i.e RealSense viewer, point cloud. I am using USB 2.0 to interface with the camera.

Here is the output.
------------------------------------------
There are 2 connected RealSense devices.

Using device 0, an Intel RealSense D415
    Serial number: 816612060691
    Firmware version: 05.09.14.00

unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
unsupported descriptor subtype: 3
Segmentation fault: 11
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

Process completed
---------------------------------------

From what I gather looking at the program the error occurs when the device pipeline is created.

RealSense-Customer-Engineering, 2018-07-31:
[Realsense Customer Engineering Team Comment]
hi @jlbeebout,

are you connecting 2 camera to PC USB2 port intentionally?
since other examples are working i think this should be issue of support formats under USB2, can you please try to connect to USB3 port and see if the examples are good?

jlbeebout, 2018-07-31:
I only have one RealSense camera attached to the computer. For some reason it is displaying that there are two connected. I unfortunately don't have a computer with USB 3.0. I do have a Raspberry Pi on order. The project I am working on requires USB 2.0 and the C examples to work together. I am testing it right now on my Mac before I try and use the C distance example on the Pi.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
hi @jlbeebout,

i see, so do you always see 2 camera connected on your Mac?
please make sure when fail, the selected camera is same as other sample that works.
another thinking is the resolution for USB2 is limited, some samples are specified one resolution only, so this maybe the root cause.

RealSense-Customer-Engineering, 2018-08-16:
[Realsense Customer Engineering Team Comment]
hi @jlbeebout,

unfortunately my Mac have only USB3 port, so i verified the sample on windows, and it's working good on D415+USB2 port, so would like to know if you load any preset before testing C samples?

RealSense-Customer-Engineering, 2018-12-05:
[Realsense Customer Engineering Team Comment]
Hi @jlbeebout,

Still need support for this topic? If not, will close this.

jlbeebout, 2018-12-05:
No I was able to get what I needed by using a different computer. Thank you!

closed 2018-12-05 by jlbeebout