realsenseai/librealsense #2176: Extracting depth information from pixels D435 realsense_camera

TAREK-ELOUARET, 2018-07-31:
Hello Everyone,

I am using D435 realsense camera, I am able to display the distance from the center of the image, but what I want to know exactlly is how to display the distance or depth information of each pixel on frame ? what's is the principal part of code which do that ? and for note, currently I am analyzing the rs-mesure code ( which has the caracteristic of measuring distance between two fixed pixels ) , but I didn't find the main part that allows me to get depth information from those pixels ? 

Thanks for your help

MartyG-RealSense, 2018-07-31:
The discussion linked to below may be of use to you.   It contains code references for getting pixel depth with just SDK 2.0, or by using SDK 2.0 in combination with the OpenCV vision software.

https://communities.intel.com/message/546117#546117

TAREK-ELOUARET, 2018-07-31:
Thanks Sir. MartyG-RealSense , It's work fine. : )

closed 2018-08-14 by RealSense-Customer-Engineering