realsenseai/librealsense #2179: Using HW Sync on D400 series with photoswitch to sync single shot capture

kyranf, 2018-08-01:
I have a D415 camera which I would like to sync a single frame capture with a photoelectric switch triggering (on a conveyor, tripped by an object).  How can I configure my D415 to listen on the HW sync pin header for the signal and to NOT take any frames (such that wait_for_frame() will not return with any data) until the signal indicates an object has arrived?

I want my image capture thread to block until a single frame is received, and that the frame exposure time is constant (with auto exposure turned off) so that I get very repeatable depth image captures of these objects on a conveyor.. 

I understand that there is new firmware available to enable the HW sync functionality on the cameras? my cameras are from the first public batch released in January 2018. 

A secondary question to this is - is there a firmware update tool which runs on Linux? I looked recently (month ago?) and saw only windows based firmware update tools. I have cameras in embedded enclosures and it's not going to be super easy to get the cable out and plug into a windows laptop to perform a firmware update. The camera is normally connected to an embedded linux micro PC.

MartyG-RealSense, 2018-08-01:
One approach that Intel took with a multi-camera capture demo this year was to "use a commercial grade flash unit to signal the start and end of a capture, allowing us to use that spike in data to make sure our camera footage was properly synchronized together".

I do not have any further technical details of how that trigger was set up.  One of the Intel guys on this GitHub can likely provide guidance on that.  

 In regard to your second question: yes, there is a Linux firmware updater tool.

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

MartyG-RealSense, 2018-08-06:
The multi-camera sync white paper says about sync triggering that whilst you can have a master and slave setup with the cameras, "Alternatively an external signal generator can also be used as the master trigger with all cameras set to slave mode.  When applying an external sync pulse, the HW SYNC input requires a 100 microsecond positive pulse at the nominal camera frame rate, 33.33 ms for a 30Hz frame rate for example".

"Inputs are high impedance, 1.8V CMOS voltage levels.  However, it is important to make sure to use a high resolution signal generator.  The frequency of the signal generator needs to exactly match the sensor frame rate.  For example, if the sensor is set up as 30 fps, the real frame rate may be 30.015 fps.  You may need to use an oscilloscope to measure the real frame and configure the signal generator to the same frequency.  For this reason, it may be better to just using one additional camera as master sync signal generator"

kyranf, 2018-08-06:
From the sounds of that, it's isn't what i'm looking for in terms of functionality :( 

I am after a SINGLE frame, with a re-trigger rate of probably 1->0.5Hz (1 second or more, before triggers).  I'll read that white paper, thank you Marty

RealSense-Customer-Engineering, 2018-12-12:
[Realsense Customer Engineering Team Comment]
Ticket being closed due to inactivity for 30+ days

closed 2018-12-12 by RealSense-Customer-Engineering

oafolabi, 2019-08-16:
@kyranf , were you able to get this to work? I am trying something similar, but the realsense doesn't seem to wait for my external signal before capturing an image. Any help is appreciated

kyranf, 2019-08-17:
@oafolabi  sorry, I haven't had the time to properly investigate this, and I was able to make my industrial part scanning just work off the leading edge detected in the depth data, referenced well enough for the robot with encoder tracking and the shared photo switch to do the job. 

I think it's really important that they add this easy 'external frame trigger' for one-shot captures, so you don't have to worry about precise timestamps or the frame jitter if it's streaming.  In my situation I just wanted it to wait until asked to take a single snapshot with an external trigger input.