realsenseai/librealsense #2180: Wrong Depth Unit

apoorva2398, 2018-08-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.13.0 }                          | 

There is a bug with the RealSense Viewer depth reported when you place the cursor over a pixel. The hexadecimal values are fine, but the multiplier is applied incorrectly when you change the depth unit. For example, 0x1462 should be displayed as 1.462 m when the depth unit is 0.001, and 0x14621 should be displayed as 1.4621 m when the depth unit is 0.0001, but the multipler is wrong (in different cases depending on D415 or D435).

dorodnic, 2018-08-02:
Hi @apoorva2398 
Yes, this is a somewhat known issue. The viewer only refreshes the depth units once. If you re-open the Viewer, you should see correct values.

dorodnic, 2018-11-06:
#2655 - better late then never

dorodnic, 2018-11-21:
Now officially released ([v2.16.5](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.5))

closed 2018-11-21 by dorodnic