realsenseai/librealsense #2181: Lowest exposure setting over-exposed image

fiorano10, 2018-08-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.09.14.00 | 
| Operating System & Version |   16.04 Ubuntu  | 
| Kernel Version (Linux Only)    |  4.15.0-24      | 
| Platform                                 | Skull canyon|
| SDK Version                            |  v2.0.3| 
| Language                            |  C++| 
| Segment			|  Robotics| 

When using the ROS wrapper, even when I have manual exposure turned all the way down to 1, the image is still over-exposed. Is there way to set the exposure setting even lower somehow? The following images were taken with auto exposure,
[IR image](https://drive.google.com/open?id=1VWIdL1g6NcMF08IpW-_k5WjWRNmjpLk_) when it's overcast 
[IR image](https://drive.google.com/open?id=1uhyNthpjl8o9yBE-BLuW9-1EUPgxrFNp) when it's sunny

MartyG-RealSense, 2018-08-02:
Have you turned off auto-exposure before setting manual exposure?  RealSense-Customer-Engineering says in a previous post on this forum that "when Autoexposure is on, the manual exposure control (and therefore UI) is not updated.  In order to get the actual exposure value the user have to read it from metadata. (frame metadata API)".

https://github.com/IntelRealSense/librealsense/issues/1624

fiorano10, 2018-08-02:
Auto-exposure was turned off before setting it manually. I'd assume if I had auto-exposure turned on it'd adjust the value to something that's not over-exposing almost the entire image.

RealSense-Customer-Engineering, 2018-08-03:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

So far, the minimal IR exposure value for D435 is 20. The solution to provide more lower exposure settings is under evaluation. Will update you if it's available.

fiorano10, 2018-08-29:
It's still not low enough for our application, we're currently using a Neutral Density filter with  region of interest auto-exposure. How badly will it affect the depth image?

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

This exposure setting in firmware is what we have implemented now. It looks your use case is very strict. You can also consider to have a actual cover to reduce the strong light issue.
Since you apply the filter to have the post processing of IR image, the depth image should have no impact.

fiorano10, 2018-09-26:
@RealSense-Customer-Engineering I ended up using an actual cover on the camera. Thanks!

closed 2018-09-26 by fiorano10