realsenseai/librealsense #2182: Firmware update tool is displaying wrong serial numbers

insensiv-ah, 2018-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 & D435 | 
| Firmware Version                   | 05.09.{09?,11}.0 -> 05.09.14.0 | 
| Operating System & Version |   Linux (Ubuntu 16.04.5 LTS)  | 
| Kernel Version (Linux Only)    |  4.4.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.14.1                          | 
| Language                            |  -                          | 
| Segment			|  others                          | 

### Issue Description
While updating the firmware of our RealSenses I've noticed that the serial number displayed by `intel-realsense-dfu` is different from the real serial number (as printed on the RealSense, packaging and reported by `rs-enumerate-devices` and `realsense-viewer`).  
This happened with each of the D415 and D435 cameras that I've updated.

If this is supposed to be the serial number of the RealSense then this is displayed wrong by `intel-realsense-dfu`.

Version of the update tool: 0.9.8

RealSense-Customer-Engineering, 2018-08-03:
[Realsense Customer Engineering Team Comment]
Hi @insensiv-ah,

I don't see this issue. What command option do you used for 'intel-realsense-dfu'?

insensiv-ah, 2018-08-03:
Hi @RealSense-Customer-Engineering,
I call the update program like this:
```
intel-realsense-dfu -b 002 -d 004 -f -i Signed_Image_UVC_5_9_14_0.bin
```
I set `-b` and `-d` according to the output of `lsusb`.  
The `-f` is probably unnecessary in most cases but in general it's the same call as given in the [DFU guide](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf).

RealSense-Customer-Engineering, 2018-08-04:
[Realsense Customer Engineering Team Comment]
Hi @insensiv-ah,

Yes, you are right. The serial number in intel-realsense-dfu is ASIC serial number, not the same optical serial number showing in realsense-viewer or rs-enumerate-devices. We should make this clearly.

RealSense-Customer-Engineering, 2018-12-22:
[Realsense Customer Engineering Team Comment]
The DFU tool is updated and the wrong S/N should not display. Please have a try.

closed 2018-12-22 by RealSense-Customer-Engineering