realsenseai/librealsense #2183: Access Pixels of RGB & IR frames on D435 realsense camera

TAREK-ELOUARET, 2018-08-02:
Hello everyone,

First of all, I would to thank's for all of you and specially who helped me on resloving my issues,  so to day I want just to ask if there is a possibility to access to each pixel of IR or RGB Image on D435 Realsense, by the way below is some of lignes codes that give you IR and RGB image in form of Mat in OpenCV:

rs2::frame color_frame = frames.get_color_frame();
rs2::frame ir_frame = frames.first(RS2_STREAM_INFRARED);

Mat color(size(640,480), CV_8UC3, (void*)color_frame.get_data(),Mat::AUTO_STEP);
Mat ir(size(640,480), CV_8UC1, (void*)ir_frame.get_data(),Mat::AUTO_STEP);

 Best refgards.
Tarek ELOUARET

TAREK-ELOUARET, 2018-08-03:
to note, I used the lignes below to display value of each channel on pixel, but it gives me false results:

for(int i=0; i<img.rows; i++)
    for(int j=0; j<img.cols; j++) 
        // You can now access the pixel value with cv::Vec3b
        std::cout << color.at<cv::Vec3b>(i,j)[0] << " " << color.at<cv::Vec3b>(i,j)[1] << " " << color.at<cv::Vec3b>(i,j)[2] << std::endl;

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Hi @TAREK-ELOUARET,

Do you try to have a configuration to set the format as follows before stream?

 //Create a configuration for configuring the pipeline with a non default profile
 rs2::config cfg;
 //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
 cfg.enable_stream(RS2_STREAM_INFRARED, 1280, 720, RS2_FORMAT_Y8, 30);
 cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);

TAREK-ELOUARET, 2018-08-13:
Hi Sir,

Thanks for your reply, but I have already solve this issue I have used the same as you gave to me. I will close this issue :  )

closed 2018-08-16 by RealSense-Customer-Engineering