realsenseai/librealsense #2186: Synchronize D415 and ZED.

Lejboelle, 2018-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Operating System & Version |  Linux Ubuntu 16  |                              | 
| Platform                                 | PC  |

Hi all,

I'm planning on doing a setup consisting of a D415 and ZED camera. Recording will be started and stopped simultaneously (connected to the same USB host) and I wish to somehow afterwards sync the two videos that have been saved.
Is this possible using any frame metadata from the D415? Any who have experienced syncing with ZED or any other non Intel cameras?
From the ZED I can get the timestamp at the time a given frame has been extracted from the USB stream.

Thanks.

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
hi @Lejboelle,

is this work for you?
do you need more support on this topic?

Lejboelle, 2018-08-14:
I've only been comparing the timestamps from a couple minutes of video, so far the TIME_OF_ARRIVAL timestamp was the one most similar to that from the ZED timestamp, being ~200ms different.
But I need to test the sync in a longer recording.

Lejboelle, 2018-08-27:
Hi,
I tried doing a longer recording, and as expected, the frames captured by ZED and RS started to drift apart when looking at the frame counter, although, it did seem that comparing the TIME_OF_ARRIVAL timestamp with the ones from ZED frames, I could sync the two outputs with ~70msec difference (~1 frame@15FPS). 
So I guess, this can be closed.

closed 2018-08-27 by Lejboelle

tugbakara, 2022-08-18:
> I tried doing a longer recording, and as expected, the frames captured by ZED and RS started to drift apart when looking at the frame counter, although, it did seem that comparing the [...]

@Lejboelle Hi, how did you make sync actually? Could you drop down here?