realsenseai/librealsense #2187: Viewer crashes when closing w/o stopping recorded sequence

ogoshen, 2018-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  development (2.14.1)                          | 

Load a recorded sequence in the viewer and while it's playing try to quit => crash (stopping the sequence before quitting doesn't)

dorodnic, 2018-08-02:
This might be a bug (I think I saw it once, but not on all systems), we will have to look more closely into it.

RealSense-Customer-Engineering, 2018-08-08:
[Realsense Customer Engineering Team Comment]
hi @ogoshen,

would like to confirm some basic for me to have a try.
what's your OS? using D415 or D435?
thanks.

ogoshen, 2018-08-13:
Windows 10, happens with recordings on either cameras.

RealSense-Customer-Engineering, 2018-08-16:
[Realsense Customer Engineering Team Comment]
hi @ogoshen,

recorded with D415 on win10, with realsense viewer from v2.15.0, then load the recorded sequence from viewer, play and click the "X" of the bag file and no issue, maybe issue fixed in v2.15, can you please confirm?

ogoshen, 2018-08-19:
'X' to close the window causes the crash. 
Closing the bag before quitting is fine.

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
hi @ogoshen,

then it's no crash here, so would like to confirm if you are testing with v2.15?
if so, can you reproduce this issue everytime? or what's the frequency?
thanks.

dorodnic, 2018-08-27:
Should be resolved in the upcoming release

closed 2018-09-03 by dorodnic