realsenseai/librealsense #2190: ROI Auto-Exposure for ROS

fiorano10, 2018-08-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.09.14.00 | 
| Operating System & Version |   16.04 Ubuntu  | 
| Kernel Version (Linux Only)    |  4.15.0-24      | 
| Platform                                 | Skull canyon|
| SDK Version                            |  v2.14.1| 
| Language                            |  C++| 
| Segment			|  Robotics| 

Is the ROI auto-exposure feature supported by ROS?

RealSense-Customer-Engineering, 2018-08-06:
[Realsense Customer Engineering Team Comment]
Hi @fiorano10,

Need to discuss internally about the feature for ROS. So far, you may need to have hard code in the ros wrapper.
Probably, you can refer to "set_region_of_interest" function in librealsense.

fiorano10, 2018-08-28:
Implemented in the ROS wrapper. Thanks!

closed 2018-08-28 by fiorano10

RaymondKirk, 2019-07-02:
> Implemented in the ROS wrapper. Thanks!

@fiorano10, @RealSense-Customer-Engineering could you provide more information about how to use/access this feature in realsense-ros?

doronhi, 2019-08-05:
> Implemented in the ROS wrapper. Thanks!

Is this option still available for you? Would you like to share the code or push a PR?

fiorano10, 2019-08-13:
@doronhi @RaymondKirk I'll create a PR soon. Thanks!

doronhi, 2019-09-03:
@fiorano10 ,
The ROI option for ROS was added in version 2.2.8 of realsense2_camera [here](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.8)
Sorry for the trouble you've gone through.

arpitg1304, 2019-12-12:
Can this be set in JSON file?

doronhi, 2019-12-15:
Not sure you can. 
You can set it with the launch file.
Why do you need it to be specifically in the JSON file?

zwn, 2020-02-19:
> You can set it with the launch file.

@doronhi Would you happen to have an example of setting the auto exposure roi in a lauch file?

doronhi, 2020-02-20:
Will that do?
```
  <rosparam>
      /camera/stereo_module/auto_exposure_roi/left: 20
      /camera/stereo_module/auto_exposure_roi/right: 200
  </rosparam>
```

Gunz-katomaran, 2023-07-13:
> Will that do?
> 
> ```
>   <rosparam>
>       /camera/stereo_module/auto_exposure_roi/left: 20
>       /camera/stereo_module/auto_exposure_roi/right: 200
>   </rosparam>
> ```

is this still in launch file?

I'm using ros noetic and D435 camera

```
[ INFO] [1689249042.382501571]: Initializing nodelet with 4 worker threads.
[ INFO] [1689249042.513004557]: RealSense ROS v2.3.2
[ INFO] [1689249042.513043439]: Built with LibRealSense v2.50.0
[ INFO] [1689249042.513070724]: Running with LibRealSense v2.50.0
[ INFO] [1689249042.555107524]:  
[ INFO] [1689249042.809256370]: Device with serial number 152122070738 was found.

[ INFO] [1689249042.811973672]: Device with physical ID 1-4-6 was found.
[ INFO] [1689249042.812172312]: Device with name Intel RealSense D435 was found.
[ INFO] [1689249042.813596997]: Device with port number 1-4 was found.
[ INFO] [1689249042.815959823]: Device USB type: 2.1
```

MartyG-RealSense, 2023-07-13:
Hi @Gunz-katomaran  Not every parameter is defined in the launch file by default, but can be added manually to the launch file with a `<rosparam> </rosparam>` code block like the one that you quoted.

Gunz-katomaran, 2023-07-14:
yeah, it's working. Thanks for your clarification.

MartyG-RealSense, 2023-07-14:
You are very welcome!