realsenseai/librealsense #2191: Install Librealsense with CUDA on Jetson Tx2

connorsoohoo, 2018-08-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 16.04}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.4.38)                                         | 
| Platform                                 | NVIDIA Jetson Tx2  |
| SDK Version                            |  { 2.14.1 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

I have installed CUDA 8.0 on the Jetson Tx2 and am trying to install Librealsense.
I can install the library fine without CUDA, but whenever I enable the CUDA flag, I am getting an issue that the CMAKE_CUDA_COMPILER flag is not found. I can compile other applications using CUDA just fine, but for some reason Librealsense cannot detect my version of CUDA.  I've been trying to manually set the CMAKE_CUDA_COMPILER flag to /usr/bin/cc but that just throws another error during build.

connorsoohoo, 2018-08-03:
Solved referring to the following issue. 

[https://github.com/jetsonhacks/buildLibrealsense2TX/issues/12](url)

closed 2018-08-03 by connorsoohoo

stutys, 2019-01-21:
@connorsoohoo hi, nothing found in this issue. 
https://github.com/jetsonhacks/buildLibrealsense2TX/issues/12

any other solutions available for this problem? I need to install librealsense with cuda on TX2.

Thank you very much