realsenseai/librealsense #2193: How to adjust depth image quality just like Depth Quality Tools in my code

trobr, 2018-08-03:
* Camera Model = SR300
 * Firmware Version  = 3.21.0.0
 * OS = Win10
 * Platform = PC
 * SDK Version = master
 
The defalut quality of depth image is very bad in my system.I find that  can adjust in Depth Quality Tools.But the configuration is lost after poweroff.So I want to know how to sync the configuration to hardware or how to adjust depth image quality just like Depth Quality Tools in my code.The following images  is the default and adjusted depth image.Thanks!
[image]
[image]

AntoineWefit, 2018-08-03:
I would advise you adjust your parameters with the **Intel RealSense Viewer** instead of the Depth Quality Tool.
You could then export these parameters in a "_.json_" file, and use this file afterward in your workflow.
[image]

dorodnic, 2018-08-03:
There might be an issue with export with the SR300, however, the SR300 performance is dominated by several key options, that you can set in any language:
```cpp
rs2::pipeline p;
auto profile = p.start();
auto dev = profile.get_device();
auto sensor = dev.query_sensors().front();
sensor.set_option(RS2_OPTION_MOTION_RANGE, ...);
sensor.set_option(RS2_OPTION_CONFIDENCE_THRESHOLD, ...);
sensor.set_option(RS2_OPTION_FILTER_OPTION, ...);
```
You could also apply one of the visual presets:
```cpp
...
sensor.set_option(RS2_OPTION_VISUAL_PRESET, RS2_SR300_VISUAL_PRESET_LONG_RANGE);
```
(see [rs2_sr300_visual_preset enum](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/h/rs_option.h#L72))

The images you attached do not necessarily indicate bad depth quality. Please note that the SR300 is a structured-light device optimized for short-range applications. We also support the D400 cameras as a more general purpose depth solution.

trobr, 2018-08-06:
@AntoineWefit Thansk!, I am very sorry for the late reply,And I can not find the option in my **Intel RealSense Viewer**.My Camera Model = SR300 and Firmware Version = 3.21.0.0.Is that because my camera doesn't support it?

trobr, 2018-08-06:
@dorodnic Thanks for your help! I am very sorry for the late reply.The camera worked fine when I followed your instructions.