realsenseai/librealsense #2195: OpCodes do not match! Sent 21 but received -6!

antodi, 2018-08-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |   legacy / 2.13.00                           | 

### Issue Description
Hi all,
The IR and depth camera suddenly stops to work but the RGB works fine. The error message I have is: 

> OpCodes do not match! Sent 21 but received -6!
[image]

[image]

As I have seen in other threads, I tried different cables, different USB ports, different computers and also a different OS (Linux). I also tried to calibrate it using the Dynamic Calibration tool but it stops at step 10, giving me the resolution. I wasn't able to go the other steps involving the target. I tried to update the firmware but it said:

> No DFU device was connected
> Failed to burn software

I also tried to update the software on Linux but I was not even able to install DFU on Ubuntu 18. 

Basically I have tried every steps on this thread: https://github.com/IntelRealSense/librealsense/issues/1508 but it seems the user had to change his whole camera. What else can I try? I am desesperate since I am using this camera for an important job with deadlines. I would appreciate any help. Thank you.

Best,
Anthony

MartyG-RealSense, 2018-08-03:
Over on the Intel Support forum for RealSense, there was a lengthy discussion about DFU upgrade failure last month.  Jesus G from Intel support suggested updating the firmware by installing the Windows Universal Windows Platform (UWP) driver, which also contains a recent firmware driver.

https://downloadcenter.intel.com/download/27799/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

I also had a suggestion about trying the Windows Troubleshooter tool.

https://communities.intel.com/message/554482#554482

antodi, 2018-08-04:
Hi MartyG,
Thank you for your message. I try the UWP but it did not solve my problem... 

>>>
[image]

The troubleshooter tool neither, unfortunately...

MartyG-RealSense, 2018-08-05:
The most common cause of this error is a failure to read the camera's calibration table, perhaps because the table is corrupted.  The common advice given is to perform a dynamic calibration.  This usually does not work when users try it though, as you yourself experienced.

Another alternative may be to use the dynamic calibration tool to reset your camera to factory default settings.

The calibration user guide says:

***********

While users calibrate devices with dynamic calibrator or through their own developed calibration tools based on Calibration API, there might be a need to read, write, and restore calibration data on the device. CustomRW Tool provides this ability to users. CustomRW Tool is included in the Calibration Tool package and is installed by default under C:\CalibrationToolAPI\2.5.2.0\bin on Windows and /usr/bin on Linux.

[image]

The –g option restores the calibration on device to default gold settings. Depending up on the type of calibration done on the device, the gold settings represents the original factory setting or the calibration setting from OEM Calibration if such a calibration is done on the device after factory calibration.

[image]

MartyG-RealSense, 2018-08-05:
Awesome news!  I'm very glad I was able to help you meet your deadline.  Good luck with your work, and I wish you a great day too. :)

closed 2018-08-14 by RealSense-Customer-Engineering