realsenseai/librealsense #2196: How can i make 2 videos simultaneously: One normal, and other with person cut out (like greenscreen)

cinemano, 2018-08-04:
How can I make 2 videos simultaneously: One normal, and other with person cut out (like greenscreen)

Can the D435 be used so that we have two videos from it recorded simultaneously?

For instance a man talking in front of a kitchen. I want both the video of him in the kitchen and him alone with everything else cut out.. How do I do this? Thanks so much...

George

MartyG-RealSense, 2018-08-04:
If you use RealSense SDK 2.0's Unity game engine implementation then there is a pre-made example that shows a depth stream, color stream and a color stream with the background segmented  out.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity

If you wish to create an application with C+ scripting (which might make recording easier to intergrate), an example program called Align can use a slider to strip away background detail from a color image.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

cinemano, 2018-08-04:
Wow.. thanks thats great. But how about also keeping the whole video as a
copy..
Idea is to have one with background segmented out but also a copy with
everything for instance to be able to mix them later in premiere.. say you
want your actor to be all blue but everyelse to be normal....

MartyG-RealSense, 2018-08-04:
If your goal is to record the full color stream and background-segmented stream to separate files simultaneously, I'm only aware of the ability to save everything to a single file when recording data into a rosbag .bag format file.

https://github.com/6RiverSystems/librealsense2/blob/master/src/media/readme.md

A hybrid approach that was suggested was to record a bag file first, and then extract a .csv format file in post-processing.

cinemano, 2018-08-04:
> blob/master/src/media/readme.md

When you say bag file, you mean a normal video and the CSV will have the
details of what to extract?
To make these tests, is the camera and a computer enough or I need the
hardware card? I have a GTX 1080

George

MartyG-RealSense, 2018-08-04:
.CSV stores tabular data such as numbers and text that can be used in spreadsheets and databases.  So that may not be the appropriate format for your project.

I researched your case carefully but could not think of an efficient way to save two color streams to file at the same time.  RealSense programming is not one of my specialist areas though, so others may be able to offer guidance.

RealSense graphics requirements are modest and do not need a dedicated graphics card.

cinemano, 2018-08-04:
Its OK.. i ended up looking at youtube videos.. its not exactly like a
greenscreen.. parts of the head appear/disapear.. was hoping it would be
far superior technology by now but were not there yet.. :/
at least in those budgets..

MartyG-RealSense, 2018-08-04:
I found details of a new product that can use D435 cameras to do green screen-like effects.

https://mixcast.me/ditching-green-screen-mixcast-gdc-2018/

cinemano, 2018-08-04:
thanks.. yes thats the camera i was looking at.. but you can see its not a
clean cut.. it waves and moves on the edges.. :/

AntoineWefit, 2018-08-06:
I don't know if that's going to help but if you haven't already seen it, you might want to take a look at this : 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/grabcuts

RealSense-Customer-Engineering, 2018-08-27:
[Realsense Customer Engineering Team Comment]
hi @cinemano,

would like to know if AntoineWefit's suggestion work for you?

RealSense-Customer-Engineering, 2018-09-17:
[Realsense Customer Engineering Team Comment]
Hi @cinemano,

Did you resolve your issue based on the sample code "grabcuts"?

closed 2018-10-08 by RealSense-Customer-Engineering