realsenseai/librealsense #2197: D435 can‘t run in arm

xiuruisi, 2018-08-06:
| Camera Model                       |Realsense D435  and  D415 | 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4,4,55-generic  | 
| Platform                                 | RP3399  |
| SDK Version                            |  librealsense2.14.1 | 
| Language                            | c++  | 

D415 can run in arm but I use USB2.0 port ,when I run D435 and it doesn't work. 

rs-multicam
  RealSense error calling rs2_ create_ device(info_ list: 0x7ab4de, index:e):
  open(...Y failed Last Error: No such file or directory）

dorodnic, 2018-08-06:
D435 / D415 both can work on any platform with USB2 / USB3. 
First, I'd recommend to check the firmware of the D435 on a different machine, using RealSense Viewer or `rs-enumerate-devices`. If the firmware is not up to date, this can cause problems. 
Next, please share `dmesg` Linux log when you connect the camera.

TAREK-ELOUARET, 2018-08-06:
Hello,

First of all, I had some problems on connecting D435 on ARM, but I realised that problems was in USB cable ( you should plug it with 3.0).

after that, try to download an old libersense version 2.10 and try to compile 

clear && cmake . && make all -j4 && sudo make all   ( be carefull to have the adequate versions of Cmake & gcc compiler to not have any issues).

After that get into Scripts folder and run the script : install_glfw3.sh 

Waiting for your response and good luck.

xiuruisi, 2018-08-06:
@dorodnic @TAREK-ELOUARET  Thank you very much. Following your methods, it works. But it occurs a new problem. When running, it's only to show the color stream and don't show the depth stream, I don't know why.

TAREK-ELOUARET, 2018-08-06:
What do you use as program or software to display ? program in examples or Realsense viewer ?

xiuruisi, 2018-08-06:
@TAREK-ELOUARET  I want to use as program and I need this depth stream.

TAREK-ELOUARET, 2018-08-06:
Go to examples/capture , and try to run rs-capture.

xiuruisi, 2018-08-07:
@TAREK-ELOUARET  I try to use this command and it doesn‘t  show the depth stream.

TAREK-ELOUARET, 2018-08-07:
@xiuruisi  Could you please display your results in image in order to get more details about your issue and thanks : )

xiuruisi, 2018-08-09:
@TAREK-ELOUARET  I'm sorry to reply to you so late. Now, the arm is the same as the original,it doesn't show the color stream and nothing shows.

TAREK-ELOUARET, 2018-08-09:
So probably, I don't think that problem is in the camera it self, could you reverify that you are using adequate cable ( USB 3.0), other wise if you could display your issue in screenshot form. and Thanks

xiuruisi, 2018-08-09:
@TAREK-ELOUARET  I also think so, these are screen captures.
[image]
[image]

The first picture is "rs-multicam" and the second is show  "lsusb" that the arm is connecting the D435(Intel Corp). I'm sure that I am using adequate cable(USB 3.0) because the arm‘s usb port is blue.

RealSense-Customer-Engineering, 2018-08-09:
[Realsense Customer Engineering Team Comment]
Hi @xiuruisi,
Please also run "lsusb -t" and "usb-devices" commands. Like to see more USB device info.

xiuruisi, 2018-08-09:
@dorodnic @TAREK-ELOUARET  These are some pictures by using command "rs-enumerate-devices", I don't know if these can help me solve problems.
[image]
[image]
[image]
[image]
[image]
[image]

xiuruisi, 2018-08-09:
Hi .@RealSense-Customer-Engineering  These are the pictures.
[image]
[image]
[image]
[image]

TAREK-ELOUARET, 2018-08-09:
After analyzing your screenshots, it's seems that you are using Litmus RTOS : ) . otherwise, as it shows above and after running "lsusb -t" command your device D435 is not connected very well ( problem of correspondence to the Board ) cause it doesn't show that Realsense D435 is connected or somthing like that. either you said that you have a bleu port in your board card, but that is not suffisent ( verify that you are using a 3.0 cable which has also a bleu output).

xiuruisi, 2018-08-09:
Now,the board card can show the image normally by using usb2.0 port  and updating the librealsense2.  I think the board has some problems and now I'm very confused about the usb ports.Thank you for your help during this time.

TAREK-ELOUARET, 2018-08-09:
You welcome Sir @xiuruisi , if you have a question in further dont hesitate to ask. and good luck : )

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Hi @xiuruisi,

Still need any further support for this topic?

xiuruisi, 2018-08-13:
Hi  @RealSense-Customer-Engineering  I don't need further support. Thank you.

closed 2018-08-13 by dorodnic