realsenseai/librealsense #2200: How to get the depth value of a specific coordinate through the stored depth picture.

njnydxzc, 2018-08-06:
I used the Intel RealSense Viewer tool in the SDK of the Intel Realsense D435 camera to store depth images. How do I get the depth value of a point of a specific coordinate of a depth image when processing a picture? For example, the point with coordinates (400,800)
[image]
[Uploading 2018.7_Depth.raw…]()

njnydxzc, 2018-08-06:
I just got in touch with this camera. I am learning about it. The question I asked may be relatively simple, but please answer a specific point.

TAREK-ELOUARET, 2018-08-08:
As I understood you want to display Depth data (distance ) of each pixel from frame_depth.

rs2::frameset frames;
frames = pipe.wait_for_frames();
// assuming that coordinates of pixel which you want extract depth is i = 600  ( for width) , j = 300 (height)

std:: cout << frame.get_distance(i, j) << endl;

Hope that it will help you, and waiting for your response.

njnydxzc, 2018-08-09:
Thank you for your answer.

However, I got some depth images that were previously taken by others using the Intel RealSense Viewer tool and stored as ".RAW" files and ".PNG" files. I want to be able to get deep data directly from these depth images instead of the RealsenseD435 device. How can I open a ".RAW" file or a ".PNG" file and get deep data?

TAREK-ELOUARET, 2018-08-09:
Oki, in this case if you have saved some frames ( images in RGB or DEPTH form) you have to open them and get processing by using OpenCV library

njnydxzc, 2018-08-09:
Thank you for your answer.

I have installed the OpenCV library. However, I was unable to find a code example related to Realsense for learning. Can you provide some relevant code examples?

Looking forward to your reply.

TAREK-ELOUARET, 2018-08-09:
Try these: 

#include <iostream>
#include <vector>
#include <opencv2/opencv.hpp>

using namespace std;
using namespace cv;

Mat frame;
frame = imread('DIRECTORY_OF_YOUR_IMAGE')

for ( int i=0; i<frame.rows; i++)
        for ( int j=0; j< frame.cols; j++)
              cout << frame.at<Vec3b>(i,j)[0]<< frame.at<Vec3b>(i,j)[1]<< frame.at<Vec3b>(i,j)[2]<< endl; // TO ACCESS EACH PIXEL FROM DEPTH OR RGB FRAME of each CHANELL

 namedWindow( "Display window", WINDOW_AUTOSIZE );// Create a window for display.
 imshow( "Display window", image );

return 0;
}

njnydxzc, 2018-08-09:
I have solved the problem I have encountered. Thank you very much for your help. 

If I have other problems, I will ask questions  on this next time.

RealSense-Customer-Engineering, 2018-08-10:
[Realsense Customer Engineering Team Comment]
@TAREK-ELOUARET
Thank you for the solution.

@njnydxzc
Glad to see the solution solves your problem, I would like to close this ticket first. In the future if you have other question, please create another new issue, if the new issue related to this one, you can link them together. Is it okay?

njnydxzc, 2018-08-13:
Sorry to bother you again.

Our project requirements require specific depth values, at least to the nearest 0.01. However, I did not find the specific depth of each color in the depth image. How can I convert the value of RGB to a specific depth value?

WeiAiFenDou, 2018-08-28:
@njnydxzc   I also encountered this problem:one people need save the depth image and others need get the deep from the depth image .How do you  solved the problem ?Please guide！ Thanks!

RealSense-Customer-Engineering, 2018-09-28:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

Since #2231 is resolved, I will close this issue.

closed 2018-09-28 by RealSense-Customer-Engineering

Vaibhavjolly, 2019-06-18:
@njnydxzc , How did u solved the problem you encountered earlier ?
 I want to get depth data directly from the depth images instead of the RealsenseD435 device. How can I open a ".RAW" file or a ".PNG" file and get depth data? In metres if possible .
Thanks in advance for the help .

Resays, 2019-06-24:
> Try these:
> 
> #include <opencv2/opencv.hpp>
> include iostream
> include vector
> 
> using namespace std;
> using namespace cv;
> 
> Mat frame;
> frame = imread('DIRECTORY_OF_YOUR_IMAGE')
> 
> for ( int i=0; i<frame.rows; i++)
> for ( int j=0; j< frame.cols; j++)
> cout << frame.at(i,j)[0]<< frame.at(i,j)[1]<< frame.at(i,j)[2]<< endl; // TO ACCESS EACH PIXEL FROM DEPTH OR RGB FRAME of each CHANELL
> 
> namedWindow( "Display window", WINDOW_AUTOSIZE );// Create a window for display.
> imshow( "Display window", image );
> 
> return 0;
> }

not working properly ... can write a code for depth raw file ?
and how to traverse depth and convert to rgb data ?