realsenseai/librealsense #2202: d435 generating "gridded" depth data.

omerbrandis, 2018-08-06:
Dear RS support,

I seem to be missing a lot of depth data using D435.
the corresponding "colorized depth image" is gridded ( uploaded image 1), and there seems to be a correlation between the missing data ( rs method get_distance returns 0 ) and the very dark pixes in the colorized depth image. 

this is happening using rs2 version 2.14 and fw 5.9.11 and 5.9.14 on d435.
but not on d415 using fw 5.9.11. ( see uploaded image 2) 
I use the exact same code to test this on my d435 and d415.

moreover, the rs-viewer does not reflect/reproduce this problem ( see uploaded image 3) , but I've not enabled post processing.
i've also exported the rs-viewers settings and used then in my code when i open the connection/streams, 
but the error persists.

please advise

Thanks,
Omer.
 
[image]
[image]
[image]

dorodnic, 2018-08-06:
Hi @omerbrandis 
This is a software issue in the `rs::align` processing block in 2.14 release. 
We are working on a fix and planning to publish in upcoming days.

Amotica, 2018-08-10:
When will this fix be applied to the pyrealsense2 in pip install?
For the mean time how can I apply this fix if I am using python on window?
Please provide a step by step guide.

dorodnic, 2018-08-12:
@omerbrandis - please take a look, this should be resolved in 2.15

omerbrandis, 2018-08-13:
Dear support,
i've now tested rs version 2.15.
the colorized image is no longer gridded, 
but it is not ok  ( it does not match the image shown in the rs-viewer)
there are dark horizonal lines  , here's a current picture 
[image]
,

please advise,
Omer.

omerbrandis, 2018-08-13:
hi,

Sorry, i've located the bug on my end,
in 2.15 colorized source is generated well.

Thanks,
Omer.

closed 2018-08-13 by omerbrandis