realsenseai/librealsense #2203: Unity Wrapper Multi Camera Point Clouds Recording

frenchtouchXR, 2018-08-06:
Hello,

I managed to live stream multiple point clouds on the 2.14.1 Unity wrapper on the same computer. But when I switch to Record mode it's impossible to save multiple streams at the same time. Only one of them is saved. It looks like the pipeline cannot save multiple streams at the same time. I get this error : 

_Exception: Camera already streaming
rs2_pipeline_start_with_config(pipe:000000004DF40820, config:000000004DF411C0)
Intel.RealSense.Helpers+ErrorMarshaler.MarshalNativeToManaged (IntPtr pNativeData)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg)
RealSenseDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:98)_

where my 98 line is : ActiveProfile = m_pipeline.Start(cfg);

Is there a way to record multiple stream ? Inside the same pipeline ? Do I need to create a new pipeline ? Is it possible ? What's your point of view ? Do you think you'll integrate the  multi rec on a next version ?

Best regards

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 x 3 } | 
| Firmware Version                   | (2.14.1) | 
| Operating System & Version |   {Win (8.1/10)} | 
                                      
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C#/unity/.NET }                          |

MartyG-RealSense, 2018-08-06:
There is an unofficial extension for the SDK 2.0 Unity wrapper that provides RGBD point clouds and multiple camera connections.  I have not used it myself and so cannot say whether it will improve your recording situation.  It may be of interest to you if you have not already seen it though.

https://communities.intel.com/message/558223#558223

frenchtouchXR, 2018-08-06:
Hello,

Thank you. I installed the Unofficial extension. It's based on an old wrapper in which the recording was not rightly included. However the package seems really interesting to calibrate multicameras. It solves partly my next issues on my project. But I still wonder how to record multicamera streams.

MartyG-RealSense, 2018-08-09:
I do not have any further insights about that.   One of the development team guys such as Dorodnic can hopefully offer advice on an approach to take.

frenchtouchXR, 2018-08-09:
I could make some tries but I'd need to know which scripts do you use on "Record" mode. 

On "Record" mode only 1 camera streams. The other ones doesn't start.

frenchtouchXR, 2018-08-10:
I managed to record multiple streams at the same time. 

I was right, the record mode couldn't recognize several real sense devices as it was written. 

I copy-pasted those 3 lines in the Realsenseconfiguration.cs 

Under the : case mode.Record :  

cfg.EnableDevice(RequestedSerialNumber);
                foreach (var p in Profiles)
                    cfg.EnableStream(p.Stream, p.StreamIndex, p.Width, p.Height, p.Format, p.Framerate);

The files are now saved. I doublechecked the records with a multi playback. It works perfectly ! 

If integrate the calibration of the unofficial extension you show me. I'll have the powerfull tool I need ! 

I feel like a sorcerer's apprentice !

closed 2018-08-10 by frenchtouchXR

MatthewLKomar, 2021-07-24:
> https://communities.intel.com/message/558223#558223

This link is dead here is where it is supposed to go to.
https://community.intel.com/t5/Items-with-no-label/Unofficial-Unity-extension-for-RealSense-400-Series/td-p/519417

MartyG-RealSense, 2021-07-24:
@MatthewLKomar   Thanks for the link update.  That old forum is obsolete and although it mostly still has its content intact, searching it typically involves going to a broken page and then using the search box to enter keywords.

Its replacement forum is at the link below.

https://support.intelrealsense.com/hc/en-us/community/topics

JulianCanoDev, 2022-09-29:
I am getting problems with new script VideoStreamRequest[] is no recognized as a datatype, I am using RealSenseSDK2.0, I would be thankful with any info.

MartyG-RealSense, 2022-09-29:
Hi @JulianCanoDev  The only case that references VideoStreamRequest[] is at https://github.com/IntelRealSense/librealsense/issues/1467

May I begin by asking the following questions, please.

1.  Which RealSense camera model are you using?

2.  Which version of RealSense SDK 2.0 are you using?

3.  If you have a Unity project, what installation method did you use to add the RealSense Unity wrapper to your Unity project?

I would recommend the following procedure:

*********

1.  Go to the RealSense SDK 'Releases' page.  Find the SDK version that you are using and go to the 'Assets' file list for that version.  Then download the Unity package file **Intel.RealSense.unitypackage** to your computer.

[image]

2.  Open your project in Unity and go to the menu option **Assets > Import Package > Custom Package**.  Navigate to the location on your computer of **Intel.RealSense.unitypackage** and select the package file to begin importing the RealSense wrapper files into your project.

JulianCanoDev, 2022-10-02:
Thanks for your answer and excuse me for my late, I fix VideoStreamRequest[] error copying VideoStreamRequest to a new script, that  contain the datatype, I installed  Intel.RealSense.unitypackage but I only can use one camera and I don't know how to use the French Script, thanks for your attention and I would happy for any info.

JulianCanoDev, 2022-10-02:
> Hi @JulianCanoDev The only case that references VideoStreamRequest[] is at #1467
> 
> May I begin by asking the following questions, please.
> 
> 1. Which RealSense camera model are you using?
> 2. Which version of RealSense SDK 2.0 are you using?
> 3. If you have a Unity project, what installation method did you use to add the RealSense Unity wrapper to your Unity project?
> 
> I would recommend the following procedure:
> 
> 1. Go to the RealSense SDK 'Releases' page.
> 
> https://github.com/IntelRealSense/librealsense/releases
> 
> 2. Find the SDK version that you are using and go to the 'Assets' file list for that version.  Then download the Unity package file **Intel.RealSense.unitypackage** to your computer.
> 
> [image]
> 
> 3. Open your project in Unity and go to the menu option **Assets > Import Package > Custom Package**.  Navigate to the location on your computer of **Intel.RealSense.unitypackage** and select the package file to begin importing the RealSense wrapper files into your project.

Thanks for your answer and excuse me for my late, I fix VideoStreamRequest[] error copying VideoStreamRequest to a new script, that contain the datatype, I installed Intel.RealSense.unitypackage but I only can use one camera and I don't know how to use the French Script, thanks for your attention and I would happy for any info.

MartyG-RealSense, 2022-10-02:
Hi @JulianCanoDev  For a multiple camera setup in the RealSense Unity wrapper, https://github.com/IntelRealSense/librealsense/issues/8347#issuecomment-778745413 advises to use multiple RsDevice components with their own unique serial number.

Expanding upon this advice, a detailed guide shared by a RealSense user for creating a Unity multicam setup is provided here:

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper