realsenseai/librealsense #2204: Convert pointcloud to Depth image (3d to 2d)

alt01, 2018-08-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Depth camera D435 | 
| Firmware Version                   | 05.09.11.00 | 
| Operating System & Version |  Windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.12.0 | 
| Language                            |  python                  | 
| Segment			|  Robot                        | 

### Issue Description

I obtained a depth image, RGB image and Point cloud (.ply) from the intel RealSense Viewer. Then I made a segmentation process in matlab, so I deleted some points of the original point cloud, but Im still having a .ply file. Now I need to convert this .ply file (the processed point cloud) into a depth image. I would like to use either matlab or python.

Do you have any idea on how to do this?.

 
Original data obtained from Intel´s SDK Viewer:

 
[image]
 
a) Original depth image (I alrready have it in gray scale).   
 
[image]

 b) Original point cloud
 
 
[image]
 
c) Processed pointcloud, in matlab (it is also a .ply file). It is the same as the original, but I deleted some points.  I need to convert this one into a depth image.

dorodnic, 2018-08-07:
Hi @alt01 
We don't have a closed processing block ready for this task, but you could use [rs2_project_point_to_pixel](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L15) to project new points to camera plane.

MartyG-RealSense, 2018-08-07:
Hi Dorodnic, this case is also on the Intel Support forums.  The user agreed with your suggestion that using rs2_project_point_to_pixel was the way to go.  I wasn't able to provide an explanation for how to do this though.

The full discussion on Intel Support is here:

https://communities.intel.com/message/558449#558449

alt01, 2018-08-07:
I read that this function needs the intrinsic paramenters of the depth camera, and I have alrready obtained them:
width: 424, height: 240, ppx: 214.063, ppy: 120.214, fx: 213.177, fy: 213.177, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0]

I dont really understand what does the""rs2_project_point_to_pixel(...)"  function return. In the documentation, is said tha it converts a 3d point into a 2d image, but is it a depth image?  and Do I have to introduce every point (x,y,z) independently? I couldn't find an example on how to do this.

dorodnic, 2018-08-07:
```cpp
float point[2];
float vertex[3] { x, y, z };
rs2_project_point_to_pixel(point, &intrinsics, vertex);
// read output pixel coordinates from point[0] and point[1]
```
You'd have to iterate over all points (x,y,z) in the point-cloud. 

There is no simple example for this use-case because this is not a common use-case, but `rs2_project_point_to_pixel` is being used in some SDK tools.

alt01, 2018-08-07:
Thanks @dorodnic .

point [0] and point[1] are the row and column of the pixel, on the depth image, corresponding to the (x,y,z) point?

alt01, 2018-08-07:
I applied sdk´s function "rs.rs2_project_point_to_pixel()" with the following inputs:

- Depth intrinsic values:   width: 424, height: 240, ppx: 214.063, ppy: 120.214, fx: 213.177, fy: 213.177, model: Brown Conrady, coeffs: [0, 0, 0, 0, 0] 

- point to convert (from pointcloud): [-0.968, -0.582, 1.03]

**The result I got is: [13.718368530273438, -0.24103546142578125]**

How should I interpret this value if its suposed to give a 2d image pixel?

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

Your result should be mapped to color pixel. I modified the python code based on #1890 to convert from depth to 2D color.

you can also refer to below article.
https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

import pyrealsense2 as rs
import numpy as np
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipeline = rs.pipeline()
pipe_profile = pipeline.start(config)
frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
# Intrinsics & Extrinsics
depth_intrin = depth_frame.profile.as_video_stream_profile().intrinsics
color_intrin = color_frame.profile.as_video_stream_profile().intrinsics
depth_to_color_extrin = depth_frame.profile.get_extrinsics_to(color_frame.profile)
color_to_depth_extrin = color_frame.profile.get_extrinsics_to(depth_frame.profile)
print("\n Depth intrinsics: " + str(depth_intrin))
print("\n Color intrinsics: " + str(color_intrin))
print("\n Depth to color extrinsics: " + str(depth_to_color_extrin))
# Depth scale - units of the values inside a depth frame, i.e how to convert the value to units of 1 meter
depth_sensor = pipe_profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("\n\t depth_scale: " + str(depth_scale))
depth_image = np.asanyarray(depth_frame.get_data())
depth_pixel = [200, 200]   # Random pixel
depth_value = depth_image[200][200]*depth_scale
print("\n\t depth_pixel@" + str(depth_pixel) + " value: " + str(depth_value) + " meter")
# from pixel to 3D point
depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin, depth_pixel, depth_value)
print("\n\t 3D depth_point: " + str(depth_point))
# from 3D depth point to 3D color point
color_point = rs.rs2_transform_point_to_point(depth_to_color_extrin, depth_point)
print("\n\t 3D color_point: " + str(color_point))
# from color point to 2D color pixel
color_pixel = rs.rs2_project_point_to_pixel(color_intrin, color_point)
print("\n\t color_pixel: " + str(color_pixel))

RealSense-Customer-Engineering, 2018-08-29:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

any update?

alt01, 2018-08-29:
The solution was to deproject each point (x,y,z) from the point cloud

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
Hi @alt01,

Is this issue resolved? You can also review #1601 to correctly convert to 2D pixel from point cloud.

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
Still need any support for this topic? If not, will close this.

closed 2018-10-24 by dorodnic

Woodstock94, 2019-05-06:
hi all, I'm struggling with the same problem here (using pyrealsense 2).
I only have the ply file (every point has x,y,z,r,g,b info).
It is clear that I can use 
`color_pixel = rs.rs2_project_point_to_pixel(color_intrin, color_point)`
to work on the single point. 
I'm able to use this function on all my points... but how can I use the output of this function in order to retrieve a real image?
I don't know how to put all the single results of the "rs2_project_point_to_pixel" function in order to build a matrix corresponding to an image. 
Could you please help me figure out this?

eyildiz-ugoe, 2019-08-09:
@RealSense-Customer-Engineering  I've ran your code but the mapped pixel point was `color_pixel: [55854.78125, 143.9385223388672]` which is ridiculous, since X cannot be that high. I am using D435 with resolutions of both sensors (color and depth) being 1280x720.

In other words, I think the code has a flaw.