realsenseai/librealsense #2205: Can't open configuration file

AleBasso80, 2018-08-07:
| Camera Model                        D400 | 
| Firmware Version                   05.09.14.00 | 
| Operating System & Version   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    4.15.0-30 | 
| Platform                                  PC |
| SDK Version                           2.14.1  | 
| Language                               RealSenseViewer / C++  | 

Hello,
I downloaded from the repos the latest SDK version and I uploaded the latest camera firmware. I'm still facing the same problem I mentioned in a previous issue when I was still using previous SDK and firmware releases.
Whenever I try to load a configuration file I saved by means of the viewer I get this error:

Unknown in rs2_load_json(dev:0x7fc01c0008c0, json_content:0x2a76800, content_size:3374):
Operation failed with error code=-6

I get it in all my PCs. Looking at the open issues, it seems it is affecting only me. I start thinking it is related to some json parsing library I install or something like that.
Do you have any idea?

I attach the configuration file I save through the viewer.

Cheers

{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
hi @AleBasso80,

i tried to reproduce this issue and load your json file with Realsense viewer 2.13.0 and it's working good.
will check with latest version tomorrow, but would like to know the error you see is right after json file loaded? or start stream after json file loaded? or else?
and i assume you are using D415?

AleBasso80, 2018-08-14:
Hi,
I'm actually using a D435 camera.
The streams do no start, as the problem arises when I try to load the config file, I assume in the parsing stage.
I tried with two different PCs and with several SDK versions, but I've never succeeded. I wonder if there is any unmet dependency for the json parsing library.

Cheers

AleBasso80, 2018-08-16:
Hello,
I don't have access to my PC for a few days. As soon as I can I make the test you suggested and I'll let you know the result.

thanks!

AleBasso80, 2018-08-29:
Hello,
I installed the packages from repo and I got the v2.16.0.
I still get the error: Unknown in rs2_load_json(dev:0x7f8fa00008c0, json_content:0x3a6e0e0, content_size:3374):
Operation failed with error code=-6

I attach the created json file:
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "24",
    "param-maxscorethreshb": "2047",
    "param-medianthreshold": "500",
    "param-minscorethresha": "1",
    "param-neighborthresh": "7",
    "param-raumine": "1",
    "param-rauminn": "1",
    "param-rauminnssum": "3",
    "param-raumins": "1",
    "param-rauminw": "1",
    "param-rauminwesum": "3",
    "param-regioncolorthresholdb": "0.0499022",
    "param-regioncolorthresholdg": "0.0499022",
    "param-regioncolorthresholdr": "0.0499022",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "10",
    "param-robbinsmonroincrement": "10",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.375",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "325",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

Thanks!

RealSense-Customer-Engineering, 2018-09-05:
[Realsense Customer Engineering Team Comment]
Hi @AleBasso80,

Suggest you can try the realsense-viewer in the pre-built packages below to see how it works?
$ sudo apt-get install librealsense2-utils
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

AleBasso80, 2018-09-05:
Hi,
I did it but I got exactly the same error. This is why I thought it could be due to some dependencies.

RealSense-Customer-Engineering, 2018-09-06:
[Realsense Customer Engineering Team Comment]
Hi @AleBasso80,

1. Did you connect to USB2 port? Is the D435 enumerated as a USB3 device? Please try "usb-devices" command and show me the log.
2. Please enable the realsense log, run "export LRS_LOG_LEVEL=DEBUG". After you load your preset file thru realsense-viewer then close the app, you should find a log file there.

AleBasso80, 2018-09-06:
[2018-09-06-14_48_32.log](https://github.com/IntelRealSense/librealsense/files/2357071/2018-09-06-14_48_32.log)

[usbdevices.txt](https://github.com/IntelRealSense/librealsense/files/2357072/usbdevices.txt)

Cheers!

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @AleBasso80,

From the logs, I can't see the obvious error related to your issue. Can you try to load our preset file (ex: HighResHighAccuracyPreset.json) in below link and see how it works?
https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets
And, if you just switch the preset in realsense-viewer, not load the preset file, it should work, right?

So far, still not sure about if it's an unmet dependency issue for the json parsing library,

AleBasso80, 2018-09-07:
Hi,
I get the same error even if I try to load the json file I found in your link: 
Unknown in rs2_load_json(dev:0x7fb6b00008c0, json_content:0x4082760, content_size:1546):
Operation failed with error code=-6

Everything is fine if I switch the preset in the realsense-viewer.

I have run out of ideas...
Thanks!

RealSense-Customer-Engineering, 2018-09-07:
[Realsense Customer Engineering Team Comment]
Hi @AleBasso80,

You can try to run below python script. Like to see it has the same error during json file loading.

# First import the library
import pyrealsense2 as rs
import time
import json

DS5_product_ids = ["0AD1", "0AD2", "0AD3", "0AD4", "0AD5", "0AF6", "0AFE", "0AFF", "0B00", "0B01", "0B03", "0B07"]
def find_device_that_supports_advanced_mode() :
    ctx = rs.context()
    ds5_dev = rs.device()
    devices = ctx.query_devices();
    for dev in devices:
        if dev.supports(rs.camera_info.product_id) and str(dev.get_info(rs.camera_info.product_id)) in DS5_product_ids:
            if dev.supports(rs.camera_info.name):
                print("Found device that supports advanced mode:", dev.get_info(rs.camera_info.name))
            return dev
    raise Exception("No device that supports advanced mode was found")
try:
    dev = find_device_that_supports_advanced_mode()
    advnc_mode = rs.rs400_advanced_mode(dev)
    print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")
    # Loop until we successfully enable advanced mode
    while not advnc_mode.is_enabled():
        print("Trying to enable advanced mode...")
        advnc_mode.toggle_advanced_mode(True)
        # At this point the device will disconnect and re-connect.
        print("Sleeping for 5 seconds...")
        time.sleep(5)
        # The 'dev' object will become invalid and we need to initialize it again
        dev = find_device_that_supports_advanced_mode()
        advnc_mode = rs.rs400_advanced_mode(dev)
        print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")

    # open Json file
    fd = open('<your json file>', 'rt')
    json_string = fd.read()
    fd.close()

    print("Controls as JSON: \n", json_string)
    # We can also load controls from a json string
    advnc_mode.load_json(json_string)
except Exception as e:
    print(e)
    pass

AleBasso80, 2018-09-10:
Hi,
I downloaded the library pip package and run your script. I didn't get any error either loading a preset file (downloaded from your link) or a json file saved through the realsense-viewer (where I get the exception if I try to load them).

What do you think it may be?
Cheers!

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
Hi @AleBasso80,

Here, I just use "pyrealsense2" module built from librealsense. The json load function included.
I guess probably, the librealsense library you built for realsense-viewer has something wrong to link the json parsing function. I also notice that your json file content size is a littler larger (ex: content_size:3374) than mine.
What else the json parsing library installed in your system? Probably, we can have a cross check.

AleBasso80, 2018-09-10:
Hi,
I have installed boost (Version: 1.58.0.1ubuntu1). I often use rapidjson but, as this is not installed (header only), I don't think it can mess things up.
Please find in attachment the output of the ldd command related to the realsense-viewer. You can see all the dynamically linked libraries.

[realsense-viewer-ldd.txt](https://github.com/IntelRealSense/librealsense/files/2365956/realsense-viewer-ldd.txt)

Thanks!

NicolasBernard456, 2018-09-18:
Hello, 

I'm having getting the same error message when loading any JSON file on the realsense viewer.
I also didn't get any issue running the small python code with custom/default JSON files.

Is there any update on this?

Thanks.

AleBasso80, 2018-10-01:
Hi,
I've always worked with Ubuntu 16.04 and I've tried different librealsense2 releases from 2.10 to 2.16. I've always got the same error. I can't say anything about other OS, as I only work in Linux.

NicolasBernard456, 2018-10-01:
Hi,
my realsense-viewer is based on librealsense 2.16.0 + ubuntu 16.04, and I'm sorry but I work also only on Linux.
I tried it on a laptop as well as on a desktop computer and I saw no changes.
Thanks for your help,
Nicolas

RealSense-Customer-Engineering, 2018-10-24:
[Realsense Customer Engineering Team Comment]
Hello AleBasso80,

We have been able to replicate similar issues with non-powered USB 3.0 hubs, or USB ports that do not deliver enough power to cameras.
Please answer the questions below so we can help narrow down your issue:

-  What kind of system are you using?  Intel UP Board, NUC, Laptop?  Please provide model name and #.

-  Are you plugging in camera to USB 3.0 ports?

-  Have you tried different USB 3.0 type-C cables and plugging those in different ports?

-  Are you using a USB 3.0 powered hub?  We only recommend powered USB 3.0 hubs, non powered devices produce errors similar to what you are seeing.

-  If you haven't already, please download and install the latest FW.

https://downloadcenter.intel.com/download/28237/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

Thank you!

AleBasso80, 2018-10-25:
Hello,
I used the USB cables provided by the Italian RealSense reseller, I'm sure I plugged the cameras to USB3 ports and I used the latest firmware version.

I tried with a laptop ASUS ROG Zephyrus and with a desktop PC (USB ports connected to the motherboard).

Cheers!

drzraf, 2018-10-31:
Same here.
Same a couple of settings. Try to load them ==> **error**
(librealsense version: 2.16.1)

RealSense-Customer-Engineering, 2018-11-02:
[Realsense Customer Engineering Team Comment]
Hello @AleBasso80 and @drzraf,

Using a USB 3.0 powered hub can possibly mitigate the issues being seen with loading .json files with RealSense cameras on RealSense-Viewer app.  It seems that not enough power is being delivered to cameras.

Before we continue to try any other steps, please consider acquiring a USB 3.0 powered hub such as this one:  https://www.amazon.com/Anker-7-Port-Adapter-Charging-iPhone/dp/B014ZQ07NE/ref=sr_1_4?ie=UTF8&qid=1541183065&sr=8-4&keywords=usb+3+powered+hub

Notice this hub as a separate power cable that will provide a 12V / 3A power adapter, not all USB 3.0 hubs provide these adapters.  This extra power is essential to deliver needed power for RealSense Depth cameras to function on low power systems / usb ports.

Please try with USB 3.0 powered hub and update this ticket if you see the same results.
Thank you.

drzraf, 2018-11-03:
Even if it would be true, the fact that `rs-viewer` fails at loading JSON because of unreliable power is a bug in itself. One would then expect "`Can not brought depth-camera up`" which is way clearer)
In my case, Realsense camera is the sole USB device connected and runs on a USB-3 port of my laptop. It's matching expected power-specs, so if it can't power depth-camera under these conditions, it's definitely a firmware bug.

ipa-jcl-dc, 2018-11-29:
Hello,
I have exactly the same problem with the newest firmware and SDK. I have tried several D435 cameras and different Ubuntu 16.04 PCs, the same problem appeared.
Does anyone know how to solve it?
Thanks.
[image]

Miaou446, 2019-01-04:
Hello, I had the same issue (D415&D435 with librealsense 2.17.0 and firmware 5.10.6)  but I've resolved it. It seems that the problem is the float numbers in the json files, the '.' separator isn't recognized. So I have replaced all the '.' by ',' and now all seems to work. Tell me if it works for you too.

RealSense-Customer-Engineering, 2019-01-05:
[Realsense Customer Engineering Team Comment]
Thank you @Miaou446 for this fix!

We have not been able to replicate this specific error, only errors similar in syntax but not exact.  Could you possibly give any other information about how you are replicating this error?

How are you generating JSON files?

This would also explain why the pyrealsense examples worked in loading JSON files.

Since this is apparently a fix, we will close this issue in 5 days if no other responses.

Thank you!

Miaou446, 2019-01-07:
Hello,

I've used the intel realsense viewer to generate the json file with the button "save current stereo module settings to file"  (realsense-viewer app on Ubuntu 16.04, librealsense 2.17.0). Then when I tried to load these settings with the button "Load pre-configured stereo module settings", I got direclty the same error as ipa-jcl-dc in his post. When I deleted the lines with float numbers in the json file I had no problems anymore so I understood that the issue was linked with the float numbers.

You're welcome

AleBasso80, 2019-01-07:
Hello,
It worked for me too. Thank you very much! 
Anyway, this is very strange issue as my OS is set to use "." as separator.

RealSenseCustomerSupport, 2019-01-24:
Hello @AleBasso80,

Glad the fix @Miaou446 worked for you.  We are looking into the cause of this, but since your issue is fixed, we will close this issue for now.

Thank you!

closed 2019-02-01 by RealSenseCustomerSupport