realsenseai/librealsense #2210: speed up split_frame

timprepscius, 2018-08-07:
the for loop in split_frame
is a low hanging fruit for optimization with multi core processors

it is a bottle neck

It would also be nice, of course, if we could just bypass all of the splitting etc.
I would rather deal with the raw streams from the device with some meta data describing what they are with the lowest amount of overhead.

Receive frame over usb, hand it to me.

I'll load it into the graphics card do all the processing there.  (yes- I see your cuda define, but I want to control the entire pipeline- and I'm personally not too thrilled about cuda lock in.)

dorodnic, 2018-08-07:
Hi @timprepscius 
What is your target architecture? 
Also, do you actually need both left and right infrared (sometimes `split_frame` will get invoked to work-around a missing patch for `L8` format)
The good thing about this library being open-source is that you can do whatever you want if you are not satisfied with what we decided to offer out of the box. 
In particular, [backend.h](https://github.com/IntelRealSense/librealsense/blob/master/src/backend.h) offers a fairly succinct cutting-point that would give you access to all the raw data without having to mess with platform specific streaming APIs. [Here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/CMakeLists.txt#L35) you can see how this can be wrapped in CMake.

dorodnic, 2018-08-21:
Hope you found everything you need. Feel free to open additional issues if you encounter any problems

closed 2018-08-21 by dorodnic