realsenseai/librealsense #2212: retrieve intrinsic/extrinsic paramters for D415 and SLAM?

mkestrada2, 2018-08-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                 | 
| Firmware Version                   | 05.09.02.00       | 
| Operating System & Version |   ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.14.13            | 
| Platform                                 | Desktop            |
| SDK Version                           |    2.14.1             | 
| Language                              |  C++                  | 
| Segment			       |  Robot                | 

### Issue Description###
Hi, I'm attempting to use the D415 camera as an rgb-d source for a SLAM algorithm ([ORB_SLAM2](https://github.com/raulmur/ORB_SLAM2)). For the algorithm, i need to provide a .yaml configuration file containing the intrinsic/extrinsic parameters of the camera so it do it's image analysis appropriately.  I've seen that you can custom calibrate the camera, but Ideally I would just be able to access the pre-configured for the camera without having to do the calibration.

Is there an easy way to access the file(s) containing the intrinsic/extrinsic parameters of the camera?

MartyG-RealSense, 2018-08-07:
There was a discussion on the Intel Support forum earlier this year about yaml files for ORB_SLAM2

https://communities.intel.com/message/543130#543130

On the same forum, intrinsics and extrinsics have been discussed during the last couple of days.  Please read downward from the comment linked to below.  RealSense programming expert JB455's advice is well worth listening to.

https://communities.intel.com/message/559221#559221

DannySauval, 2018-08-08:
I didn't checked the links from the other answer, but I used to need the same data as you. I couldn't find them so I used the calibrations tools to get the parameters, and it worked fine. Why don't you want to calibrate ? Not to calibrate on every camera you use ? Because if it is the case, depending on what you want to do, you may be able to use the same parameters.

RealSense-Customer-Engineering, 2018-08-10:
[Realsense Customer Engineering Team Comment]
Hi @DannySauval,

You can try the librealsense app "rs-enumerate-devices -c" to get the extrinsic/instrinsic parameters.

mkestrada2, 2018-08-10:
Hi @DannySauval, If we were planning to use it long term, we would go through the process of calibration, but the work is for a short term summer gig and we are mostly using the D415 in the interim while we wait for a MultiSense 7 camera to arrive. So we aren't super concerned with perfect calibration, just getting something that works reasonably well.  To that end we were able to get the appropriate values using the methods discussed in @MartyG-RealSense 's [second link above](https://communities.intel.com/message/559221#559221)

closed 2018-08-10 by mkestrada2