realsenseai/librealsense #2214: duplicate frame numbers in rosbag

marcovs, 2018-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.09.09.02 | 
| Operating System & Version |   MacOS  | 
| Kernel Version (Linux Only)    |                                   | 
| Platform                                 | Mac |
| SDK Version                            |  2.11.0                         | 
| Language                            | C++                        | 
| Segment			|  AR                   | 

From the output of the rosbag viewer, I noticed that sometimes consecutive frames are recorded with the same frame number. The timestamp and the frame contents are different, but the frame number is the same. The issue does not bother me, but I thought to flag it just in case it was not documented.

dorodnic, 2018-08-08:
Hi @marcovs 
Same frame number with different timestamp is not a valid behavior IMHO, we will try to reproduce and fix. 
Any specific steps you can share for obtaining such rosbag?

marcovs, 2018-08-09:
Did not doing anything different than usual, captured the rosbag with only rgb and depth streams, no infrared at 30fps with a resolution of 1280x720. I am wondering whether the fact that I am connecting the realsense through a usb hub to the macbook pro is what is causing the issue

RealSense-Customer-Engineering, 2018-08-15:
[Realsense Customer Engineering Team Comment]
hi @marcovs,

i see you are using libRealsense v2.11.0, which is kind of old now, would like to know if this can be reproduced with last release?
and how you record the bag file? via realsense viewer?
thanks.

dorodnic, 2018-09-25:
Should be resolved in [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

RealSense-Customer-Engineering, 2018-10-02:
[Realsense Customer Engineering Team Comment]
Hi @marcovs,

How about the test result based on v2.16.1?

closed 2018-10-11 by dorodnic