realsenseai/librealsense #2215: rosbag reader

marcovs, 2018-08-08:
Starting from issue #1919, I worked on it a little bit and I made a class that reads bag files using directly the rosbag API. It implements the following functionalities

1. Reads a bag file and outputs a pair of rgb and depth frames asynchronously whenever the function nextFrame is called. The class does not use the realsense pipeline, rather it parses the rosbag messages, so there are no frames dropped

2. Since AFAIK there is no hardware synchronization between rgb and depth streams on the device, the class checks the timestamps and outputs the pair of rgb and depth files that are closest in time (without expensive searches).

3. The depth frame returned is aligned to the camera point of view

4. Depth frame is saved as an xml with the full floating point values in meters for each pixel

5. Together with the frames, the class outputs the frame number and the timestamp for each frame recorded in the rosbag (according to the system clock)

6. It also saves the intrinsic and extrinsic matrices for both sensors in case you want to do some geometry

The code should be quite readable, and the dependencies are opencv 3.x, glog, librealsense, and the part of the ros library included as headers, and boost. The files print_helpers.cpp/h are a modified version of the ones included in the rosbag viewer tool example.

Even if you don't use the class directly, you should be able to copy and paste the methods inside the class to read the rosbag however you want.

Hope it helps
[RealSenseHelper.zip](https://github.com/IntelRealSense/librealsense/files/2270056/RealSenseHelper.zip)
.

dorodnic, 2018-08-08:
Hi @marcovs 
Thank you for sharing your work! This can help us solve existing problems in the SDK and help other users in meantime. 
If you prefer not to create a PR to the SDK, please add one of the [common licenses](https://opensource.org/licenses) to the archive so people can legally use it.

marcovs, 2018-08-09:
Hi @dorodnic thank you, just updated the archive with MIT license, everyone should be free to use it and hack it

dorodnic, 2018-09-25:
Hi @marcovs 
I have added the link to this page under [Examples / Community Projects](https://github.com/IntelRealSense/librealsense/blob/master/examples/readme.md). 
Feel free to reconsider making it a pull-request or relocating to GitHub repo. 
Very cool, thank you!

RealSense-Customer-Engineering, 2018-10-16:
[Realsense Customer Engineering Team Comment]
This issue should be resolved in 2.16.1.

closed 2018-10-16 by RealSense-Customer-Engineering

HippoEug, 2018-11-21:
The rosbag library (dependency) is not supported for windows, correct?

SunshineDou, 2018-11-29:
Hi， I want to use the RealsenseHealper in Visual Studio, But when I compiled, it said that:
fatal error C1083: cannot open file:"rosbag/bag.h": no such file or directory. 
what's the problem? how to solve it? @marcovs

timsta613, 2018-12-04:
Hi,

I'm having a similar problem in that I'm having trouble with pointing the project to its respective dependencies. @marcovs Is there an example CMakesList.txt file that you used for this?

hrdkjain, 2019-03-13:
There seems to be a couple of fixes to resolve include and linker error, try adding following include directories
```
include_directories(librealsense/third-party/realsense-file/rosbag/cpp_common/include/)
include_directories(librealsense/third-party/realsense-file/rosbag/rostime/include/)
include_directories(librealsense/third-party/realsense-file/rosbag/roscpp_traits/include/)
include_directories(librealsense/third-party/realsense-file/boost/)
include_directories(librealsense/third-party/realsense-file/rosbag/roscpp_serialization/include/)
include_directories(librealsense/third-party/realsense-file/rosbag/msgs/)
```
along with the `-lglog` linker flag in CMakeLists.txt
Also it was required to modify `ros::Time` to `rs2rosinternal::Time` in `CRealSenseHelper.cpp`

95ImadEL, 2019-10-14:
@marcovs @hrdkjain please could you provide us the CMakesList.txt. Thank you in advance

arothenberg, 2020-07-17:
Like some of the users I wasn't sure which   headers files and "boost" files to use from the rosbag repository. So I went to the rosbag site to just build it from scratch and do a basic tutorial.
I started  downloading  ROS and  it's dependencies from this site
http://wiki.ros.org/Installation/Windows.  It's a long download using a admin powershell.. It's several gigs I think - it's very involved. lots of stuff being installed on my pc. Mostly python. I stopped it after 6 gigs.  

I'll hack through it and hopefully be able to use this solution if that's what I need to do. No problem, but i'd like to know if I'm going about this correctly. ie  Do I need to be installing all these python programs through the "chocolatey" program that ROS had me download or is there an easier way to do this?  

Is the marcov solution currently the best way to consume bag files? Is there a simpler method that uses another format (not bag) ? I have no problem trying to make this work but I want to make sure this is the right path because it could take a bit?

Thanks.