realsenseai/librealsense #2216: Frame Drops When Recording Bag File in RealSense Viewer

apoorva2398, 2018-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435 | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.13.0                          | 

### Issue Description
When I record even just one stream as a bag file from RealSense Viewer and [open it in MATLAB](https://communities.intel.com/message/553535#553535), I find that the number of frames recorded is less than the correct amount—e.g. a 5.6 s recording at 25 fps should have 5.6 × 25 = 140 frames, but the bag file contains only 106 frames, even though clicking the information button in the Viewer shows a constant 25.00 fps (after enabling hardware metadata). The problem is worse when multiple streams are recorded, even though we use an NVMe SSD.

dorodnic, 2018-08-08:
Are you using the [pre-compiled Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.14.1/Intel.RealSense.Viewer.exe)?

apoorva2398, 2018-08-08:
I tried it with that now, but it recorded only 1082 frames for 1280 × 800 Y16 IR1 on D435 for 60 s at 25 fps (ideally, 25 × 60 = 1500 frames).

MartyG-RealSense, 2018-08-08:
For reference, below is a link to a similar case on the Intel Support forums from last week with apparent frame drops from recordings when used with MATLAB.

https://communities.intel.com/message/559327#559327

apoorva2398, 2018-08-08:
Thanks for the reference, @MartyG-RealSense, but I don't think MATLAB is the issue. (I would get the same frame drops on opening the bag file any other way.) I think it is something to do with recording in the RealSense Viewer.

For reference, here are the frame numbers I get from the bag file (which match what is seen in the Viewer). The frames are recorded correctly up to frame number 700, after which only every alternate frame seems to be recorded in general.
Info: D415, 1920 × 1080, 25 fps, IR2, Y16
Expected number of frames = 60 s × 25 fps = 1500 frames
Actual number = 842 frames

676
677
678
679
680
681
682
683
684
685
686
687
688
689
690
691
692
693
694
695
696
697
698
699
700
701
703
705
706
708
710
711
713
715
716
718
720
721
723
725
727
729
731
733
735
737
739
741
743
745
746
748
750
752
754
756
758
760
762
764
766
768
769
771
773
775
776
778
780
781
783
785
786
788
790
791
793
795
796
798
800
802
804
806
808
810
812
814
816
817
819
821
823
825
827
829
831
833
835
837
839
841
842
844
846
847
849
851
852
854
856
858
859
861
862
864
866
868
869
871
873
875
877
879
880
882
884
886
888
890
892
894
896
898
900
902
904
906
908
910
912
914
916
917
919
921
922
924
926
928
929
931
932
934
936
937
939
941
942
944
946
947
949
951
952
954
956
958
960
962
964
966
968
969
971
973
975
977
979
981
983
985
987
989
991
992
994
996
997
999
1001
1002
1004
1006
1007
1009
1011
1012
1014
1016
1017
1019
1021
1022
1024
1026
1027
1029
1031
1033
1035
1037
1038
1040
1042
1044
1046
1048
1050
1052
1054
1056
1058
1060
1062
1063
1065
1067
1069
1070
1072
1074
1075
1077
1079
1080
1082
1084
1085
1087
1089
1090
1092
1094
1095
1097
1099
1100
1102
1104
1105
1107
1109
1111
1113
1114
1116
1118
1120
1122
1124
1126
1128
1130
1132
1134
1136
1138
1139
1141
1143
1144
1146
1148
1150
1151
1153
1155
1156
1158
1160
1161
1163
1165
1167
1168
1170
1172
1173
1175
1177
1178
1180
1182
1183
1185
1187
1189
1190
1193
1194
1196
1198
1200
1202
1204
1206
1208
1209
1211
1213
1215
1217
1219
1220
1222
1224
1226
1227
1229
1231
1233
1235
1237
1239
1241
1243
1244
1246
1248
1250
1252
1254
1256
1258
1259
1262
1263
1265
1267
1269
1271
1273
1275
1277
1279
1280
1282
1284
1286
1288
1290
1292
1294
1295
1297
1299
1300
1302
1304
1306
1308
1310
1312
1313
1315
1317
1319
1321
1323
1325
1327
1329
1331
1333
1334
1336
1338
1340
1342
1344
1346
1348
1350
1352
1353
1355
1357
1359
1361
1363
1365
1367
1369
1370
1372
1374
1375
1377
1379
1380
1382
1384
1386
1388
1390
1392
1394
1396
1398
1400
1402
1404
1405
1407
1409
1411
1413
1415
1417
1419
1421
1423
1424
1426
1428
1430
1432
1434
1436
1438
1440
1442
1443
1445
1447
1448
1450
1452
1453
1455
1457
1459
1460
1462
1464
1466
1468
1470
1472
1474
1476
1478
1480
1482
1484
1486
1488
1489
1491
1493
1495
1497
1499
1501
1503
1505
1507
1509
1511
1513
1515
1517
1519
1520
1522
1524
1526
1527
1529
1531
1532
1534
1536
1538
1540
1542
1543
1545
1547
1549
1551
1553
1555
1557
1559
1561
1563
1565
1567
1569
1571
1572
1574
1576
1578
1580
1582
1584
1586
1588
1589
1591
1593
1595
1597
1598
1600
1602
1604
1605
1607
1609
1610
1612
1614
1616
1618
1620
1622
1624
1626
1628
1630
1632
1633
1635
1637
1639
1641
1643
1645
1647
1649
1650
1652
1654
1656
1658
1660
1662
1664
1666
1667
1669
1671
1672
1674
1676
1677
1679
1681
1682
1684
1686
1688
1689
1691
1693
1695
1697
1698
1700
1702
1704
1706
1708
1710
1712
1713
1715
1717
1719
1721
1723
1725
1727
1729
1731
1732
1734
1736
1738
1740
1742
1744
1745
1747
1749
1750
1752
1754
1755
1757
1759
1760
1762
1764
1765
1767
1769
1771
1773
1775
1776
1778
1780
1782
1784
1786
1788
1789
1791
1793
1795
1797
1799
1801
1803
1805
1807
1809
1810
1812
1814
1816
1818
1820
1821
1823
1825
1826
1828
1830
1831
1833
1835
1836
1838
1840
1841
1843
1845
1847
1849
1851
1853
1855
1857
1858
1860
1862
1865
1866
1868
1870
1872
1874
1876
1878
1880
1882
1883
1885
1887
1889
1891
1893
1894
1896
1897
1899
1901
1902
1904
1906
1908
1909
1911
1912
1914
1916
1918
1919
1921
1922
1924
1926
1928
1930
1932
1934
1935
1937
1939
1941
1943
1945
1947
1949
1951
1953
1954
1956
1958
1960
1962
1964
1966
1968
1969
1971
1973
1974
1976
1978
1980
1981
1983
1984
1986
1988
1989
1991
1993
1994
1996
1998
1999
2001
2003
2005
2007
2009
2011
2013
2015
2017
2018
2020
2022
2024
2026
2028
2030
2032
2033
2035
2037
2039
2041
2043
2045
2046
2048
2050
2051
2053
2055
2056
2058
2059
2061
2063
2064
2066
2068
2069
2071
2073
2075
2076
2078
2080
2082
2084
2086
2087
2089
2091
2093
2095
2097
2099
2101
2103
2104
2106
2108
2110
2112
2114
2116
2118
2120
2121
2123
2125
2126
2128
2130
2131
2133
2134
2136
2138
2139
2141
2143
2144
2146
2148
2150
2151
2153
2155
2156
2158
2160
2162
2164
2166
2168
2169
2171
2173
2175
2177
2179
2181
2183
2185

apoorva2398, 2018-08-08:
Thanks for the reference, @MartyG-RealSense, but I don't think MATLAB is the issue. (I would get the same frame drops on opening the bag file any other way.) I think it is something to do with recording in the RealSense Viewer.

For reference, here are the frame numbers I get from the bag file (which match what is seen in the Viewer). The frames are recorded correctly up to frame number 701, after which only every alternate frame seems to be recorded in general.
Info: D415, 1920 × 1080, 25 fps, IR2, Y16
Link: http://www.mediafire.com/file/ub8he4kl5nal3z1/20180808_120219.bag/file
Expected number of frames = 60 s × 25 fps = 1500 frames
Actual number = 842 frames

S. No.) Frame #
1) 676
2) 677
3) 678
4) 679
5) 680
6) 681
7) 682
8) 683
9) 684
10) 685
11) 686
12) 687
13) 688
14) 689
15) 690
16) 691
17) 692
18) 693
19) 694
20) 695
21) 696
22) 697
23) 698
24) 699
25) 700
26) 701
27) 703
28) 705
29) 706
30) 708
31) 710
32) 711
33) 713
34) 715
35) 716
36) 718
37) 720
38) 721
39) 723
40) 725
41) 727
42) 729
43) 731
44) 733
45) 735
46) 737
47) 739
48) 741
49) 743
50) 745
51) 746
52) 748
53) 750
54) 752
55) 754
56) 756
57) 758
58) 760
59) 762
60) 764
61) 766
[... 756 lines cut ...]
818) 2141
819) 2143
820) 2144
821) 2146
822) 2148
823) 2150
824) 2151
825) 2153
826) 2155
827) 2156
828) 2158
829) 2160
830) 2162
831) 2164
832) 2166
833) 2168
834) 2169
835) 2171
836) 2173
837) 2175
838) 2177
839) 2179
840) 2181
841) 2183
842) 2185

dorodnic, 2018-09-25:
Most likely this is resolved now in  [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1)

RealSense-Customer-Engineering, 2018-10-12:
[Realsense Customer Engineering Team Comment]
Hi @apoorva2398,

Any update? If no further issue, will close this.

RealSense-Customer-Engineering, 2018-10-18:
[Realsense Customer Engineering Team Comment]
Hi @upxela,

I don't see this issue when I use the realsense-viewer to record a ros-bag file and check total frame number from the bag file is expected based on the recording time. Do you also use the realsense-viewer in v2.16.1 to record and what resolution used? Test it on Ubuntu or Win10?

upxela, 2018-10-18:
@RealSense-Customer-Engineering 

This is using realsense-viewer in v2.16.1, recording 1280x720 video at 30fps, on Windows 10.

RealSense-Customer-Engineering, 2018-10-19:
[Realsense Customer Engineering Team Comment]
Hi @upxela,

1. Did you observe the varying FPS or stable FPS during recording? You can check this stream info (click the "I" mark in the stream view) for each frame in the realsense-viewer GUI.
2. You can also check "Frame#" in the stream info or use "get_frame_number()" to see if the frame arrives normally based on 30 FPS.

upxela, 2018-10-19:
@RealSense-Customer-Engineering 

I just recorded approximately 15 seconds of video with the realsense-viewer. Playing back the bag file while checking stream info shows approximately 30 fps (sometimes slightly less, e.g. 29.7). However, the number of img messages is clearly less than 450 (there are only 237 frames). 

Do you know where this discrepancy comes from?

alizarghami, 2019-01-01:
@RealSense-Customer-Engineering
Hi.
I'm facing the same problem of not getting the true frame rate when reading frames from .bag files. I had posted another issue [here](https://github.com/IntelRealSense/librealsense/issues/2411) that I think it's caused because of the same reason I'm getting lower frame rate. I was told that the problem will be solved in [v2.16.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.16.1) but I'm still facing this problem.
Here is my specification

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.16.1 | 
| Language                               | python  | 
| Segment			         | others | 

I hope we can find a solution for this problem.

realsens01, 2019-01-16:
I'm not sure if my problem is the same, but I thought it best to try posting here before creating a new issue.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5_11_1 | 
| Operating System & Version |   Windows 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.17.0 and 2.17.1 |
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description

We have used Realsense Viewer to record bag files in the past, without problems. After weeks of trying to debug a separate issue, however, we noticed that the video feeds had become choppy while recording (but not while just streaming). This is a consistent effect, and seems to happen every time we record now. When we looked at the data files, we found dropped frames were frequent, with some gaps up to 5 seconds long. I have spent a lot of time attempting to resolve this since we noticed the issue last week. Here is what I have tried:

* Uninstalled and reinstalled the SDK. Version 2.17.0.
* Built the SDK from scratch. Version 2.17.1.
* Upgraded the firmware. v5_11_1.
* Upgraded the Nvidia graphics drivers to the latest available.
* Updated Windows.
* Tested 4 different D435 cameras, individually.
* Used different USB ports, 1 of which is on a PCIe expansion card with a separate hub.
* Used Windows Device Manager to remove all previously installed drivers for Realsense.
* Tried different camera stream combinations and settings.
* Checked resource usage during recording (CPU around 30%, 34 MB/s disk IO / 23% of highest active time, 50-60% of physical memory in use).
* Etc.

I just tested recording with a python script, and there were no gaps to be found. A 1 minute recording in Realsense Viewer yielded a 2GB bag file. A 1 minute recording with the python script yielded a 4GB bag file. I saved both bags as well as videos of what the recordings looked like on playback (they have the same duration, as expected, but the Realsense Viewer bag file playback is choppy... as it was in real-time). I will send those if they are useful. I also have Windows usbview log files that I exported. I'm attaching the Realsense Viewer log file taken during the 1 minute of recording. That was at INFO log level, so I am also attaching another log file for a quick recording with DEBUG log level.

```python
import time, pyrealsense2 as rs2

if __name__ == '__main__':
    device        = rs2.context().devices[0]
    serial_number = device.get_info(rs2.camera_info.serial_number)
    config        = rs2.config()
    pipeline      = rs2.pipeline()
    config.enable_device(serial_number)
    config.enable_all_streams()
    config.enable_record_to_file('test.bag')
    pipeline.start(config)
    time.sleep(60)
    pipeline.stop()
    del pipeline
```

It's probably important to note that metadata is disabled, as that seems to be a critical factor pertaining to our other issue (described at great length on our [community post](https://forums.intel.com/s/question/0D50P0000490Z6WSAU/frame-errors-when-recording-from-multiple-cameras-on-a-single-computer), and to a lesser extent as part a [github issue](https://github.com/IntelRealSense/librealsense/issues/2754)).

[realsense_viewer-debug_log_level-2019-01-15.log](https://github.com/IntelRealSense/librealsense/files/2762699/realsense_viewer-debug_log_level-2019-01-15.log)
[realsense_viewer-1_min_recording-2019-01-15.log](https://github.com/IntelRealSense/librealsense/files/2762700/realsense_viewer-1_min_recording-2019-01-15.log)

TheMikeyR, 2019-02-11:
I have the same issue as @realsens01 

Any information regarding this issue @MartyG-RealSense  @dorodnic ?

We have tried it on three different pc's 

All software versions are 2.18.0
Same camera have been tried, same usb cables. It seems the issue is regarding compression, when turning compression off in the realsense-viewer the camera works just fine. 

**Compression on recording**
*PC 1 with lags*
On the PC's which lags while recording, when compression is turned on (which is device default i guess since it is also an issue here)
It reports 35 MB/s disk write
Realsense Viewer use 130% CPU when not recording
When recording turns on it uses 180% and maxes out one core to 100%

*PC 2 with no lags*
79 MB/s disk write
120% CPU usage on realsense viewer
60% max on one core

**Compression off recording**
*PC 1 with lags*
Disk Write 130 MB/s
Realsense Viewer use 130% CPU when not recording
Realsense viewer uses 147% CPU and one core max out at 73% CPU.  

*PC 2 with no lags*
130 MB/s disk write
120% CPU usage on realsense viewer
60% max on one core

**Recording specifications**
Depth at 30 fps in 1280x720 with Z16
RGB at 30 fps in 1280x720 with RGB8

**PC 1 (Lags)**
Ubuntu 16.04 with kernel 4.15.0-45-generic
CUDA 10.0
Nvidia Driver Version: 410.48
Tried both compiled and from apt  

**PC 2 (Works fine)**
Ubuntu 16.04 with kernel 4.15.0-45-generic
Cuda 9.0
Nvidia Driver Version: 390.87
Installed from apt

**PC 3 (Lags)**
Windows 10
Tried pre-compiled

MartyG-RealSense, 2019-02-11:
The link below discusses how some PCs may have problems with using compression because the CPU does not handle the built-in frame compression.

https://github.com/IntelRealSense/librealsense/issues/2102#issuecomment-436881193

TheMikeyR, 2019-02-11:
I've just realized after updating my apt so realsense should be v2.18.1 that I was running my compiled version, after deleting all compiled source code and reinstalling from apt, I can run it without any lag and disk write is around 79 MB/s. 

So it works from apt sources, but when compiling it from source it lags on Ubuntu

TheMikeyR, 2019-02-11:
Also PC 1 is running a m1200 quadro with i7-7700hq
PC 2 is running m1000 quadro with i7-6700hq
PC 3 is the same as PC2

So the issue should not be due to low-end systems as in #2102

RealSenseCustomerSupport, 2019-02-27:
----------------------------------------------

Hi @TheMikeyR,

Did you build realsense-viewer 2.18.1 as Release Build and has the issue only when the compression enabled and depth+RGB records with the default configuration? And, the firmware is 5.11.1.0?

TheMikeyR, 2019-03-01:
I didn't specify to build it as a realease.
I created a build folder and used `cmake ../ -DBUILD_EXAMPLES=true`

The issue only occurs with compression enabled (or left as default which I assume is compression on).  

The depth+rgb record is set to 1920x1080 for rgb and 1280x720 for depth at 30 fps. 

Tried with both firmware 5.11.1.0 and 5.10.13

RealSenseCustomerSupport, 2019-03-06:
----------------------------------------------

Hi @TheMikeyR,

The issue you reported can be observed. If any update, will let you know.

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

This depends on the system computing/process power (CPU, RAM, storage, SW stack, etc) in order to avoid the frame drops.

realsens01, 2019-04-02:
Then why does recording function properly through the SDK but not through Realsense Viewer? Why did @TheMikeyR observe the issue when compiling from source but not via apt?

Can you elaborate? It has to be more than just hardware restrictions if some changes to the software can resolve the issue, right? Am I mistaken?

closed 2019-04-09 by RealSenseCustomerSupport