realsenseai/librealsense #2218: RS2_FRAME_METADATA_ACTUAL_EXPOSURE can't be retrieved

dledents, 2018-08-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | 5.9.14.0 | 
| Operating System & Version |   {Win (8.1/10)   | 
| Platform                                 | PC  |
| SDK Version                            |  2.13.0.18                         | 
| Language                            |  {C }                          | 

### Issue Description
I would like to get an actual exposure value in case when  the AutoExposure is enabled.
As it has been suggested here: [https://github.com/IntelRealSense/librealsense/issues/1624](url), the value could be retrieved via RS2_FRAME_METADATA_ACTUAL_EXPOSURE attribute:

> frameset = pipe.wait_for_frames();
 std::cout << "meta exposure = " << frameset.get_depth_frame().get_frame_metadata(rs2_frame_metadata_value::RS2_FRAME_METADATA_ACTUAL_EXPOSURE) << std::endl;

But when I try to request RS2_FRAME_METADATA_ACTUAL_EXPOSURE value, I get an exception that "metadata not available".
Is it possible to get exposure value for D415 sensor?

ev-mp, 2018-08-09:
Hello @dledents,
Metadata attributes are advanced capabilities and require additional preparation steps that are specified in the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md).

With metadata enabled in the registry and activated in the code you should be able to obtain actual exposure values for D415.

dledents, 2018-08-10:
Hello @ev-mp 
Thank you for answer. I followed the steps from the installation guide. But I can't figured out how to activate metadata in the code.
Does it work in Windows 8.1?

dledents, 2018-08-10:
I can get get exposure value for depth and color frames on Windows 10. But the exposure value for the color frame never changes. Is it expected?

ev-mp, 2018-08-13:
Hello @dledents ,

The actual exposure metadata attribute for RGB sensor updates only with the "Auto-exposure" flag is "Off". Unfortunately this is a HW limitation of the sensor.

Windows OS supplied Metadata configuration and access API with Win10 release, hence Win10 is a [prerequisite](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows) for Metadata handling in Librealsense.

dledents, 2018-08-14:
Ok, got it. So I am closing this issue.

closed 2018-08-14 by dledents