realsenseai/librealsense #2223: Cannot save depth frames using rs2::frame::keep()

heavyman1, 2018-08-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 05.09.13.00                                             |
| Operating System & Version |   Ubuntu 16.04.1                                     | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                  | 
| Platform                                 | Up Board                                                |
| SDK Version                            |  2.12.0                                                    | 
| Language                               |  C++                                                       | 
| Segment                                |  others                                                    | 

I want to save frames and then post-processing them.
I found below samples from issue #1942.
Color frames are saved successfully, but depth frames are not.
Depth frames are saved only 4 frames, other 96 frames are same with 4th depth frame.

Code : 

    #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
    #include <iostream>

    int main(int argc, char * argv[]) try
    {
        const int N = 100;
        std::vector<rs2::frame> colors, depths;

        rs2::config config;
        config.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
        config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
        rs2::pipeline pipe;
        pipe.start();

        for (int i = 0; i < N; i++)
        {
            auto fs = pipe.wait_for_frames();
            
            auto depth = fs.get_depth_frame();
            auto color = fs.get_color_frame();
            
            depth.keep();
            color.keep();
            
            depths.push_back(depth);
            colors.push_back(color);
        }
        
        double preTimestampDepth = depths[0].get_timestamp();
        double preTimestampColor = colors[0].get_timestamp();
        for (int i = 0; i < N; i++)
        {
            double timestampDepth = depths[i].get_timestamp();
            double timestampColor = colors[i].get_timestamp();
            printf("frame [%02d]. depth : frame# %03d, timestamp diff [%7.3f]. color : frame# %03d, timestamp diff[%7.3f].\n", i, 
                depths[i].get_frame_number(), timestampDepth - preTimestampDepth,
                colors[i].get_frame_number(), timestampColor - preTimestampColor);
            preTimestampDepth = timestampDepth;
            preTimestampColor = timestampColor;
        }

[... 89 lines cut ...]
frame [75]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 076, timestamp diff[ 33.610].
frame [76]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 077, timestamp diff[ 33.597].
frame [77]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 078, timestamp diff[ 33.728].
frame [78]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 079, timestamp diff[ 33.572].
frame [79]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 080, timestamp diff[ 33.456].
frame [80]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 081, timestamp diff[ 33.760].
frame [81]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 082, timestamp diff[ 33.593].
frame [82]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 083, timestamp diff[ 33.451].
frame [83]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 084, timestamp diff[ 33.592].
frame [84]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 085, timestamp diff[ 33.600].
frame [85]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 086, timestamp diff[ 33.865].
frame [86]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 087, timestamp diff[ 33.484].
frame [87]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 088, timestamp diff[ 33.607].
frame [88]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 089, timestamp diff[ 33.596].
frame [89]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 090, timestamp diff[ 33.581].
frame [90]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 091, timestamp diff[ 33.619].
frame [91]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 092, timestamp diff[ 33.593].
frame [92]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 093, timestamp diff[ 33.604].
frame [93]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 094, timestamp diff[ 33.643].
frame [94]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 095, timestamp diff[ 33.529].
frame [95]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 096, timestamp diff[ 33.629].
frame [96]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 097, timestamp diff[ 33.482].
frame [97]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 098, timestamp diff[ 33.716].
frame [98]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 099, timestamp diff[ 33.480].
frame [99]. depth : frame# 010, timestamp diff [  0.000]. color : frame# 100, timestamp diff[ 33.671].

heavyman1, 2018-08-13:
Camera is connected via USB3. Video from both sensors are stable.

The code snippet in issue #1942, you keep frame after post-processing.
But I do not apply post-processing filters to the depth frame from frameset.
I want to save frames, and then applying post-processing filters after acquiring frames are finished.

When I do not apply post-processing filter to the depth frame from frameset, I can not acquire new depth frame.

    int main()
    {
        try
        {
            const int N = 100;
            std::vector<rs2::frame> colors, depths;

            rs2::decimation_filter dec_filter;
            rs2::spatial_filter spat_filter;
            rs2::temporal_filter temp_filter;
            rs2::disparity_transform depth_to_disparity(true);
            rs2::disparity_transform disparity_to_depth(false);

            rs2::config config;
            config.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 30);
            config.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_RGB8, 30);
            rs2::pipeline pipe;
            pipe.start(config);

            for (int i = 0; i < N; i++)
            {
                auto fs = pipe.wait_for_frames();
                
                auto depth = fs.get_depth_frame();
                auto color = fs.get_color_frame();

                rs2::frame filtered = std::move(depth);

                // filtered = dec_filter.process(filtered);
                // filtered = depth_to_disparity.process(filtered);
                // filtered = spat_filter.process(filtered);
                // filtered = temp_filter.process(filtered);
                // filtered = disparity_to_depth.process(filtered);

                filtered.keep();
                color.keep();
                
                depths.push_back(filtered);
                colors.push_back(color);
            }
            
            double preTimestampDepth = depths[0].get_timestamp();
            double preTimestampColor = colors[0].get_timestamp();
            for (int i = 0; i < N; i++)
            {
                double timestampDepth = depths[i].get_timestamp();
                double timestampColor = colors[i].get_timestamp();
                printf("frame [%02d]. depth : frame# %03lld, timestamp diff [%7.3f]. color : frame# %03lld, timestamp diff[%7.3f].\n", i, 
                    depths[i].get_frame_number(), timestampDepth - preTimestampDepth,
                    colors[i].get_frame_number(), timestampColor - preTimestampColor);
                preTimestampDepth = timestampDepth;
                preTimestampColor = timestampColor;
            }
        }
        catch (const rs2::error & e)
        {
            std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
        }
        catch (const std::exception& e)
        {
            std::cerr << e.what() << std::endl;
        }
[... 249 lines cut ...]
    frame [78]. depth : frame# 123, timestamp diff [ 33.275]. color : frame# 116, timestamp diff[ 32.295].
    frame [79]. depth : frame# 125, timestamp diff [ 66.724]. color : frame# 118, timestamp diff[ 78.540].
    frame [80]. depth : frame# 126, timestamp diff [ 33.295]. color : frame# 119, timestamp diff[ 34.092].
    frame [81]. depth : frame# 128, timestamp diff [ 66.680]. color : frame# 121, timestamp diff[ 66.409].
    frame [82]. depth : frame# 129, timestamp diff [ 33.386]. color : frame# 122, timestamp diff[ 33.980].
    frame [83]. depth : frame# 131, timestamp diff [ 66.683]. color : frame# 124, timestamp diff[ 65.124].
    frame [84]. depth : frame# 132, timestamp diff [ 33.361]. color : frame# 125, timestamp diff[ 32.844].
    frame [85]. depth : frame# 134, timestamp diff [ 66.604]. color : frame# 127, timestamp diff[ 64.857].
    frame [86]. depth : frame# 135, timestamp diff [ 33.375]. color : frame# 128, timestamp diff[ 33.693].
    frame [87]. depth : frame# 137, timestamp diff [ 66.681]. color : frame# 130, timestamp diff[ 65.056].
    frame [88]. depth : frame# 138, timestamp diff [ 33.332]. color : frame# 131, timestamp diff[ 33.619].
    frame [89]. depth : frame# 140, timestamp diff [ 66.772]. color : frame# 133, timestamp diff[ 65.892].
    frame [90]. depth : frame# 141, timestamp diff [ 33.292]. color : frame# 134, timestamp diff[ 33.903].
    frame [91]. depth : frame# 143, timestamp diff [ 66.652]. color : frame# 136, timestamp diff[ 67.289].
    frame [92]. depth : frame# 144, timestamp diff [ 33.356]. color : frame# 137, timestamp diff[ 33.882].
    frame [93]. depth : frame# 146, timestamp diff [ 66.755]. color : frame# 139, timestamp diff[ 64.936].
    frame [94]. depth : frame# 147, timestamp diff [ 33.457]. color : frame# 140, timestamp diff[ 34.197].
    frame [95]. depth : frame# 149, timestamp diff [ 66.513]. color : frame# 142, timestamp diff[ 65.741].
    frame [96]. depth : frame# 150, timestamp diff [ 33.508]. color : frame# 143, timestamp diff[ 35.449].
    frame [97]. depth : frame# 152, timestamp diff [ 66.535]. color : frame# 145, timestamp diff[ 64.814].
    frame [98]. depth : frame# 153, timestamp diff [ 33.296]. color : frame# 146, timestamp diff[ 33.997].
    frame [99]. depth : frame# 155, timestamp diff [ 66.687]. color : frame# 148, timestamp diff[ 67.584].

But when I apply post-processing filters before saving frames, frame rate is dropped from 30fps to about 20fps.
D435 works as 30fps, but I cannot get next frame within 33ms due to post-pocessing filters.

heavyman1, 2018-08-17:
How can i keep depth frame without post-processing filters?

RealSense-Customer-Engineering, 2018-08-22:
[Realsense Customer Engineering Team Comment]
Hi @heavyman1,

I tried your code (without post-processing) based on librs 2.15.0 + FW5.9.14 and didn't see your issue.
What's your system test configuration?

RealSense-Customer-Engineering, 2018-08-29:
[Realsense Customer Engineering Team Comment]
Hi @heavyman1,

Let me know if the issue still there?

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
Hi @heavyman1,

Do you have any update result based on the latest librealsense and firmware?

RealSense-Customer-Engineering, 2018-09-19:
[Realsense Customer Engineering Team Comment]
Hi @heavyman1,

any update?

closed 2018-11-09 by RealSense-Customer-Engineering