realsenseai/librealsense #2224: Frame number of depth frame and color frame does mismatch on linux

heavyman1, 2018-08-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 05.09.13.00                                             |
| Operating System & Version |   Ubuntu 16.04.1                                     | 
| Kernel Version (Linux Only)    |  4.13.0-36-generic                                  | 
| Platform                                 | Up Board                                                |
| SDK Version                            |  2.12.0                                                    | 
| Language                               |  C++                                                       | 
| Segment                                |  others                                                    | 

Frame number of depth frame and color frame does not match on linux.

Code : 

    #include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
    #include <iostream>

    int main(int argc, char * argv[])
    {
        try
        {
            const int N = 100;

            rs2::config config;
            config.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_RGB8, 30);
            config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
            rs2::pipeline pipe;
            pipe.start();

            double preTimestampDepth = 0;
            double preTimestampColor = 0;
            for (int i = 0; i < N; i++)
            {
                auto fs = pipe.wait_for_frames();
                auto depth_frame = fs.get_depth_frame();
                auto color_frame = fs.get_color_frame();
                double timestampDepth = depth_frame.get_timestamp();
                double timestampColor = color_frame.get_timestamp();
                if (i == 0)
                {
                    printf("frame [%02d]. depth : frame# %lld, timestamp diff [%7.3f]. color : frame# %lld, timestamp diff[%7.3f].\n", i, 
                        depth_frame.get_frame_number(), 0.0f,
                        color_frame.get_frame_number(), 0.0f);
                }
                else
                {
                    printf("frame [%02d]. depth : frame# %lld, timestamp diff [%7.3f]. color : frame# %lld, timestamp diff[%7.3f].\n", i, 
                        depth_frame.get_frame_number(), timestampDepth - preTimestampDepth,
                        color_frame.get_frame_number(), timestampColor - preTimestampColor);
                }
                preTimestampDepth = timestampDepth;
                preTimestampColor = timestampColor;
            }

            return EXIT_SUCCESS;
        }
        catch (const rs2::error & e)
        {
[... 86 lines cut ...]
frame [75]. depth : frame# 79, timestamp diff [ 33.392]. color : frame# 76, timestamp diff[ 33.643].
frame [76]. depth : frame# 80, timestamp diff [ 33.322]. color : frame# 77, timestamp diff[ 33.593].
frame [77]. depth : frame# 81, timestamp diff [ 33.255]. color : frame# 78, timestamp diff[ 33.659].
frame [78]. depth : frame# 82, timestamp diff [ 33.310]. color : frame# 79, timestamp diff[ 33.511].
frame [79]. depth : frame# 83, timestamp diff [ 33.388]. color : frame# 80, timestamp diff[ 33.606].
frame [80]. depth : frame# 84, timestamp diff [ 33.295]. color : frame# 81, timestamp diff[ 33.654].
frame [81]. depth : frame# 85, timestamp diff [ 33.339]. color : frame# 82, timestamp diff[ 33.558].
frame [82]. depth : frame# 86, timestamp diff [ 33.283]. color : frame# 83, timestamp diff[ 33.660].
frame [83]. depth : frame# 87, timestamp diff [ 33.410]. color : frame# 84, timestamp diff[ 33.551].
frame [84]. depth : frame# 88, timestamp diff [ 33.244]. color : frame# 85, timestamp diff[ 33.646].
frame [85]. depth : frame# 89, timestamp diff [ 33.372]. color : frame# 86, timestamp diff[ 33.538].
frame [86]. depth : frame# 90, timestamp diff [ 33.269]. color : frame# 87, timestamp diff[ 33.602].
frame [87]. depth : frame# 91, timestamp diff [ 33.374]. color : frame# 88, timestamp diff[ 33.572].
frame [88]. depth : frame# 92, timestamp diff [ 33.229]. color : frame# 89, timestamp diff[ 33.622].
frame [89]. depth : frame# 93, timestamp diff [ 33.428]. color : frame# 90, timestamp diff[ 33.597].
frame [90]. depth : frame# 94, timestamp diff [ 33.299]. color : frame# 91, timestamp diff[ 33.664].
frame [91]. depth : frame# 95, timestamp diff [ 33.287]. color : frame# 91, timestamp diff[  0.000].
frame [92]. depth : frame# 96, timestamp diff [ 33.329]. color : frame# 92, timestamp diff[ 33.600].
frame [93]. depth : frame# 97, timestamp diff [ 33.306]. color : frame# 93, timestamp diff[ 33.584].
frame [94]. depth : frame# 98, timestamp diff [ 33.264]. color : frame# 94, timestamp diff[ 33.634].
frame [95]. depth : frame# 99, timestamp diff [ 33.463]. color : frame# 95, timestamp diff[ 35.515].
frame [96]. depth : frame# 100, timestamp diff [ 33.172]. color : frame# 96, timestamp diff[ 31.668].
frame [97]. depth : frame# 101, timestamp diff [ 33.309]. color : frame# 97, timestamp diff[ 33.495].
frame [98]. depth : frame# 102, timestamp diff [ 33.372]. color : frame# 98, timestamp diff[ 33.705].
frame [99]. depth : frame# 103, timestamp diff [ 33.288]. color : frame# 99, timestamp diff[ 33.466].

dorodnic, 2018-08-12:
Hi @heavyman1 
I think this can be caused by auto-exposure.
Can you try replacing `pipe.start(config)` in your code with:
```cpp
auto profile = pipe.start(config);
auto color = profile.get_device().query_sensors()[1];
color.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);
color.set_option(RS2_OPTION_EXPOSURE, 100);
```

heavyman1, 2018-08-13:
Result is same.
But even if frame number is different with depth and color, frames from same frameset are acquired at the same time?

dorodnic, 2018-08-13:
Here are the rules for Depth-RGB sync:
a. Depth and RGB have to be in same actual FPS. This can be forced by disabling `exposure_priority` control (this way auto-exposure will not reduce FPS).
b. Depth frame and RGB frame get their timestamps in different parts of the (hardware)pipe. I believe depth in start frame and RGB mid exposure. This means that visually the frames should be the same, but TS will never be exactly the same. 

Also, it is a good idea to check `frame.get_timestamp_domain()` - this should be `hardware_clock`.

heavyman1, 2018-08-13:
frame timestamp domain is system time.
how can I set timestamp domain to hardware_clock?

RealSense-Customer-Engineering, 2018-09-03:
[Realsense Customer Engineering Team Comment]
Hi @heavyman1,

any update?

closed 2018-09-10 by RealSense-Customer-Engineering