realsenseai/librealsense #2226: D435 inconsistently Allowing 1080p RGB camera

rofoto, 2018-08-12:
Hi, 

   I've been using the D435 camera for a few weeks now and am finding that sometimes ( seemingly randomly ) the option for 1080p is unavailable. I am running win10 . I am on the newest firmware ( 05.09.14.00 ) and have kept the camera on its own usb3 connection. 

   This is quite a big issue as I cannot test if I cannot reliably get the camera to send the same resolution feed between launches. 

Any help would be great. 

Thank you

rofoto, 2018-08-13:
> [Realsense Customer Engineering Team Comment]
hi @rofoto,

i would like to reproduce this issue so need some basic from you:

what's your OS?
which SDK version you are using?
can this issue reproduce with Realsense viewer? like restart the viewer or need to reboot every time.
thanks.

I am on windows 10 
Realsense 2.0 ( latest available downloads ) 
yes this issue is repeatable in the viewer app. In fact it sometimes persists on reboots and full shutdown / startup cycles.

RealSense-Customer-Engineering, 2018-08-16:
[Realsense Customer Engineering Team Comment]
hi @rofoto,

while i am testing, i wonder this maybe cable quality issue that you connect to USB3 port but sometimes recognized as USB2 device? though we can support 720p x 15fps under USB2.

RealSense-Customer-Engineering, 2018-08-16:
[Realsense Customer Engineering Team Comment]
correction, for RGB on USB2, we don't support 720p@15FPS, so yes, this seems more like somehow the device was recognized as USB2 device, please help to check when fail, your device is recognized as USB2 or not,
thanks.

rofoto, 2018-08-19:
@RealSense-Customer-Engineering 

As it stands now I have not had this issue come up in a few days. I have experienced this issue using the packaged cable and switched to a USB3 powered extension cable and have not had issues since. You may be correct. It would be helpful to have the error message describe power as a potential cause to avoid having this pop up again. 

Thanks for the reply

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
hi Rofoto,

good to hear you have no issue now, cable quality issue is not something we can control, but for your case, you can detect if the camera is connected as USB2 to avoid this kind of issue.

timprepscius, 2018-08-21:
I'd like to tag along here too.

Is there a "full reset" function which can be used on startup?  Which will be immune the current state of the device.

I am seeing that if I kill the program using the device, on the next run, symttoms described by rofoto occur.  My full reset currently is unplugging the device and plugging it back in.  ;-/ . I think saw that a full reset function in the sdk somewhere.

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
hi @timprepscius,

for reset, i think you can refer to issue 2161, FYR.

closed 2018-12-12 by RealSense-Customer-Engineering