realsenseai/librealsense #2227: Unity instructions generate x86 version, Unity requires x64 version

bleppie, 2018-08-12:
Following the instructions to generate the librealsense2.sln Visual Studio project creates projects for x86. Trying to open/run the unity samples after a successful compilation doesn't work because it expects the plugins to have been built in x64. Adding an additional option of 
-DCMAKE_GENERATOR_PLATFORM=x64 to the instructions for cmake fixes the issue.

Please fix the documentation.

This is the same as issue #2031 which was closed even though the problem still exists.

RealSense-Customer-Engineering, 2018-08-13:
[Realsense Customer Engineering Team Comment]
Thanks, @bieppie,

I have created a PR as follows to update the instructions.
https://github.com/IntelRealSense/librealsense/pull/2228

dorodnic, 2018-08-15:
@bleppie thank you for pointing this out. The new instructions will be part of next release.

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
Hi @bleppie,

The instructions for Unity updated. Will close this.

closed 2018-08-20 by RealSense-Customer-Engineering