realsenseai/librealsense #2229: Changing disparity shift in c++

MechanicJay7, 2018-08-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |5.9.14 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  LibRealsense SDK 2.0                         | 
| Language                            | c++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello I'm trying to change disparity shift via code in my project but couldn't find an efficient way so far.
What I want to do is to change disparity shift to the desired value such as 100 if needed.
I know that disparityShift parameter is in the structure STDepthTableControl but I cannot figure out a way to get access to this structure in my project and modify that value so that i can get depth stream with modified diaparityShift value.

Thanks

dorodnic, 2018-08-13:
Something like:
```cpp
using namespace rs2;
context ctx;
auto dev = ctx.query_devices().front().as<advanced_mode>();
STDepthTableControl table = dev.get_depth_table();
// modify table
dev.set_depth_table(table); // write changes
```
If you are using pipeline:
```cpp
pipeline p;
auto profile = p.start();
auto dev = profile.get_device().as<advanced_mode>();
// same as before
```

MechanicJay7, 2018-08-13:
@dorodnic  It works well and I really appreciate your help, thanks.

One more question, Is there any way that I can handle presets AND the small manipulation like disparity shift at the same time?

dorodnic, 2018-08-13:
Most presets will reset all advanced mode controls, but then you can change the controls back. 
You can also [download the preset you want in JSON form](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) and remove the parts you wish to control yourself. In general, switching presets during streaming (without stop / start) is not recommended (not all registers can be modified without stop/start). In the Viewer we require stop / start because of it.

dorodnic, 2018-08-21:
@MechanicJay7 
Does this answer your question?

dorodnic, 2018-08-30:
Feel free to open more issues if you have more questions.

closed 2018-08-30 by dorodnic