realsenseai/librealsense #2231: How to get the depth value corresponding to the color in the depth image?

njnydxzc, 2018-08-13:
I got some depth images that were previously taken by others using the Intel RealSense Viewer tool and stored as ".RAW" files and ".PNG" files. I want to be able to get depth data directly from these depth images instead of the Intel Realsense D435 device.Our project requirements require specific depth values, at least to the nearest 0.01.  How can I open a ".RAW" file or a ".PNG" file and get depth data?Or how to get the depth value corresponding to the color in the depth image?

njnydxzc, 2018-08-13:
Three days ago, I mentioned this question, but it was not completely solved. See 
https://github.com/IntelRealSense/librealsense/issues/2200

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

You can refer to "rs-align" example. And, use below code to get the depth value in the aligned color. Hope this is helpful.

  const int w_other = other_frame.as<rs2::video_frame>().get_width();
  const int h_other = other_frame.as<rs2::video_frame>().get_height();
  cv::Mat otherMat(cv::Size(w_other, h_other), CV_8UC3, (void*)other_frame.get_data(), cv::Mat::AUTO_STEP);
  const int w_aligned = aligned_depth_frame.as<rs2::video_frame>().get_width();
  const int h_aligned = aligned_depth_frame.as<rs2::video_frame>().get_height();
  cv::Mat alignMat(cv::Size(w_aligned, h_aligned), CV_16U, (void*)aligned_depth_frame.get_data(), cv::Mat::AUTO_STEP);
  imshow("AlignFrame", alignMat);
  .....
  To get the frame center's depth: alignMat.at<ushort>(w_aligned/2, h_aligned/2)

ev-mp, 2018-08-14:
Hello @njnydxzc , 
The ".RAW" data is just a blob of the data transmitted over USB. The data layout is according to the [UVC spec for the depth stream](https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/pixfmt-z16.html) . It is intended for machine vision and tools such as matlab/opencv , as suggested in  #2200.
Depth ".PNG" provides a colorized version of the depth suitable for human representation. Since the colorizer uses histogram equalization It is not possibly to extract depth data directly.

As for the other question, in order to create depth<->rgb correspondence librealsense provides "align" processing block (see `rs-align` demo). 
In case you only have recorded depth and color in raw/bin formats then you'll still be able to use the "align" APIs but you'll have to write code to perform it manually:
Prerequisites:
Depth frames in ".RAW/.BIN" format + stream intrinsic.
Color frames in RGB/PNG format + stream intrinsic.
Depth<->Color extrinsic.
A sync map between Depth to Color frames that can be based on frame id or timestamp.

Flow:
1. Create "rs2::software_device"(`rs-software-device` demo).
2. Add synthetic streams for depth and color including  intrinsic/extrinsic data.
3. Create and feed a synchronized pair of depth/color to synthetic streams with data obtained from ".RAW"/".PNG" files.
4. Use the synthetic frames as inputs to "align" block to produce "A aligned to B" frame.
5. Repeat [3,4] for all pairs of depth/color

njnydxzc, 2018-08-14:
Sorry, I didn't find the code of "rs-align" example in the SDK.
Which header files do I need to use to run this code?

RealSense-Customer-Engineering, 2018-08-14:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

The rs-align sample code in below link. And, you need to include the opencv header like "#include <opencv2/opencv.hpp>" if you like to use above code.
https://github.com/IntelRealSense/librealsense/blob/53537d4a5115d0abbd79e5379eaa93d701344e4f/examples/align/readme.md

njnydxzc, 2018-08-15:
Depth data is available through the Realsense D435 device. Why can't I get deep data when I open a picture with OpenCV? Is it because the depth data is not saved?

njnydxzc, 2018-08-15:
Depth data is available through the Realsense D435 device. Why can't I get deep data when I open a image with OpenCV? Is it because the depth data is not saved?

njnydxzc, 2018-08-15:
Thank you for your reply.@RealSense Customer Engineering
If I don't have a Realsense D435 device connected, how do I define the types of  other_frame and aligned_depth_frame in the code  and load the image?

RealSense-Customer-Engineering, 2018-08-17:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

My way is for active streams from D435. How about @ev-mp's suggestions to handle the RAW and PNG data?
Or, you can have the RGB and depth aligned, and then have the record and playback as the example "record-playback".

njnydxzc, 2018-08-17:
Thank you for your reply.@RealSense Customer Engineering
Our project wants to use a deep learning algorithm, so we will collect the images in advance and then process them.So, I want to know if there is a way to get deep data from a saved image without D435.

RealSense-Customer-Engineering, 2018-08-22:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

Suggest to try @ev-mp's align flow if the saved PNG/RAW files used. And, RAW file has the depth data.

njnydxzc, 2018-08-23:
Is there a function that can be used directly to get deep data in a ".RAW" file?

RealSense-Customer-Engineering, 2018-08-29:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

Since the RAW data layout is according to the UVC spec for the depth stream, you can try below sample code to get the 16bit depth data:

FILE *fp = NULL;
int framesize = IMAGE_WIDTH * IMAGE_HEIGHT;
fp = fopen(fileName, "rb");
imagedata = (char*)malloc(sizeof(short) * framesize);
fread(imagedata, sizeof(short), framesize, fp);
cv::Mat depthMat(Size(IMAGE_WIDTH, IMAGE_HEIGHT), CV_16U, (void*)imagedata, cv::Mat::AUTO_STEP);
// show the depth frame central value from the raw data
printf("Depth RAW Central Value: %d mm", depthMat.at<ushort>(IMAGE_HEIGHT / 2, IMAGE_WIDTH / 2));

RealSense-Customer-Engineering, 2018-09-10:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

still need any support for this topic?

njnydxzc, 2018-09-20:
Thank you for your reply.@RealSense Customer Engineering

When I run the code,“ depthMat.at(IMAGE_HEIGHT / 2, IMAGE_WIDTH / 2)" reports an error.
Which header files do I need to use to run this code?

x-jeff, 2018-09-20:
"depthMat.at<uchar>(IMAGE_HEIGHT / 2, IMAGE_WIDTH / 2)" is ok .

njnydxzc, 2018-09-20:
Thank you for your reply.@x-jeff.

Which header files do I need to use to run this code?

x-jeff, 2018-09-20:
`#define _CRT_SECURE_NO_WARNINGS
#include<iostream>
#include<opencv2/opencv.hpp>;
using namespace std;
using namespace cv;
int main(){
	int IMAGE_WIDTH = 424, IMAGE_HEIGHT = 240;
	FILE* fp;
	char* imagedata;
	int framesize = IMAGE_WIDTH*IMAGE_HEIGHT;
	fp = fopen("C:/personal document/SegmentationAlgorithm/grabcut/GrabcutTest/record/frame1/d_Depth.raw", "rb");
	imagedata = (char*)malloc(sizeof(short)* framesize);
	fread(imagedata, sizeof(char), framesize, fp);
	cv::Mat depthMat(Size(IMAGE_WIDTH, IMAGE_HEIGHT), CV_8U, (void*)imagedata, cv::Mat::AUTO_STEP);
	printf("Depth RAW Central Value: %d mm", depthMat.at<uchar>(IMAGE_HEIGHT / 2, IMAGE_WIDTH / 2));
	fclose(fp);
	free(imagedata);
	return 0;
}`
This is my code, the code can run normally, you can refer to it.

njnydxzc, 2018-09-21:
Thank you for your reply.@x-jeff.

I got some data from the code, but it doesn't seem to be very accurate. Does the data need to be extra processed to make it more accurate?

RealSense-Customer-Engineering, 2018-09-21:
[Realsense Customer Engineering Team Comment]
Hi @njnydxzc,

you mean your depth data from RAW is not accurate? How is its accuracy error? ex: more than 2% error within 2m?

njnydxzc, 2018-09-21:
@RealSense Customer Engineering
I first got deep data through the "Intel RealSense Viewer" tool. Then, I save the depth image and get the depth data of the same coordinate through the code. The two data are not the same.

njnydxzc, 2018-09-22:
@x-jeff @RealSense Customer Engineering
I have solved the problem. Thank you very much for your help.
If I have other problems, I will  create another new issue.

closed 2018-09-25 by RealSense-Customer-Engineering

moonCheng, 2018-12-28:
> @x-jeff @RealSense Customer Engineering
> I have solved the problem. Thank you very much for your help.
> If I have other problems, I will create another new issue.

Hello, I  have the same problem as you, could you tell me how you solved it in the end？

L-xn, 2022-05-21:
> 【Realsense客户工程团队点评】 你好@njnydxzc,
> 
> 你的意思是你的RAW深度数据不准确？它的精度误差如何？例如：2m 内误差超过 2%？

There are errors in my depth image. I filtered the depth flow, but I could not extract the depth flow after filtering.  
There is an error in the program.
`align_to = rs.stream.color
    align = rs.align(align_to)

    # Streaming loop
    while True:
        # Get frameset of depth
        frames = pipeline.wait_for_frames()
        aligened_frames=align.process(frames)
        aligened_depth_frame=aligened_frames.get_depth_frame()

        color_frame=frames.get_color_frame()

        spatial=rs.spatial_filter()
        spatial_depth=spatial.process(aligened_depth_frame)

        temporal = rs.temporal_filter()
        temporal.set_option(rs.option.filter_smooth_alpha, 0.1)
        temporal.set_option(rs.option.filter_smooth_delta, 20)
        tem_depth = temporal.process(spatial_depth)

        hole_filling = rs.hole_filling_filter()
        filled_depth = hole_filling.process(aligened_depth_frame)

        # Colorize depth frame to jet colormap
        depth_color_frame = colorizer.colorize(filled_depth)

        # Convert depth_frame to numpy array to render image in opencv
        depth_color_image = np.asanyarray(depth_color_frame.get_data())
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_color_image, alpha=0.41), cv2.COLORMAP_JET)
        color_image=np.asanyarray(color_frame.get_data())

        w=filled_depth.get_width()
        h=filled_depth.get_height()

        dis=round(filled_depth.get_distance(int(w/2),int(h/2))*100,2)
        print("the dis are",dis,"cm away")
[image]
`