realsenseai/librealsense #2232: .NET encapsulation Sensor m_instance

chris-wojciechowski, 2018-08-13:
In recent comming for .NET wrapper m_instance field of Sensor class had it's visibility changed which makes it impossible to access methods not yet available through the wrapper and slows down R&D for our team. Could this be reviewed or is there another way of finding the sensor pointer that can be used in the native methods?

dorodnic, 2018-08-13:
Seems like it was changed from `public` to `internal`, which sounds like a good idea.
We could add a public getter to help your use case. You can also add it in form of a pull-request (this will ensure it will make it into the next release)

chris-wojciechowski, 2018-08-13:
Thank you, I will create a pull-request.

dorodnic, 2018-08-31:
Should be resolved now

closed 2018-08-31 by dorodnic