realsenseai/librealsense #2237: point cloud data wrapping/tiling

wongfei, 2018-08-14:
point cloud data is wrapped on sides, any way to fix or cut these regions?

[image]

[image]

`void ARealSenseInspector::UpdatePointCloud()
{
	SCOPED_PROFILER;

	const int NumPoints = RsPoints->size();
	if (NumPoints <= 0) return;

	const int NumPclPoints = (int)(NumPoints * PclDensity);
	const int Step = FMath::FloorToInt(NumPoints / (float)NumPclPoints);
	if (!NumPclPoints || !Step) return;

	REALSENSE_TRACE(TEXT("PCL id=%d Points=%d PclPoints=%d Vertices=%d Step=%d"), 
		PclFramesetId, NumPoints, NumPclPoints, NumPclPoints * 4, Step);

	const rs2::vertex* SrcVertices = RsPoints->get_vertices();
	const rs2::texture_coordinate* SrcTexcoords = RsPoints->get_texture_coordinates();

	{
		NAMED_PROFILER("_AllocMesh");
		PclVertices.SetNumUninitialized(NumPclPoints * 4); // 4 vertex per point
		PclIndices.SetNumUninitialized(NumPclPoints * 6); // 6 indices per point
	}

	{
		NAMED_PROFILER("_GenerateMesh");
		FPointCloudVertex* DstVertices = &PclVertices[0];
		int32* DstIndices = &PclIndices[0];
		FPointCloudVertex PV[4];

		const float Size = (PclVoxelSize * 0.5f);
		int PointId = 0;
		int VertexId = 0;
		int IndexId = 0;

		for (int i = 0; i < NumPclPoints; i++)
		{
			const auto V = SrcVertices[PointId];
			const auto T = SrcTexcoords[PointId];

			// the positive x-axis points to the right
			// the positive y-axis points down
			// the positive z-axis points forward
			const FVector Pos = FVector(V.z, V.x, -V.y) * 100.0f * PclScale; // meters to mm

			PV[0].Position = FVector(Pos.X, Pos.Y - Size, Pos.Z - Size);
			PV[1].Position = FVector(Pos.X, Pos.Y - Size, Pos.Z + Size);
			PV[2].Position = FVector(Pos.X, Pos.Y + Size, Pos.Z + Size);
			PV[3].Position = FVector(Pos.X, Pos.Y + Size, Pos.Z - Size);

			PV[0].Normal = FVector(-1, 0, 0);
			PV[1].Normal = FVector(-1, 0, 0);
			PV[2].Normal = FVector(-1, 0, 0);
			PV[3].Normal = FVector(-1, 0, 0);

			PV[0].Tangent = FVector(0, 0, 1);
			PV[1].Tangent = FVector(0, 0, 1);
			PV[2].Tangent = FVector(0, 0, 1);
			PV[3].Tangent = FVector(0, 0, 1);

			PV[0].UV0 = FVector2D(T.u, T.v);
			PV[1].UV0 = FVector2D(T.u, T.v);
			PV[2].UV0 = FVector2D(T.u, T.v);
			PV[3].UV0 = FVector2D(T.u, T.v);

			DstVertices[0] = PV[0];
			DstVertices[1] = PV[1];
			DstVertices[2] = PV[2];
			DstVertices[3] = PV[3];

			DstIndices[0] = VertexId + 0;
			DstIndices[1] = VertexId + 2;
			DstIndices[2] = VertexId + 1;
			DstIndices[3] = VertexId + 0;
			DstIndices[4] = VertexId + 3;
			DstIndices[5] = VertexId + 2;

			DstVertices += 4;
			DstIndices += 6;
			VertexId += 4;
			IndexId += 6;
			PointId += Step;
		}
	}

	{
		NAMED_PROFILER("_UpdateMeshSection");
		PclMesh->UpdateMeshSection(0, PclVertices, PclIndices); // TODO: SLOW AS HELL
	}
}
`

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
Hi @wongfei,

The point cloud tiling on sides can be reproduced. Will further check this internally to see any solution there.

wongfei, 2018-08-23:
Also point cloud tex coords are different when using align depth to color.

sample:
[image]

not aligned:
[image]

aligned:
[image]

RealSense-Customer-Engineering, 2018-09-11:
[Realsense Customer Engineering Team Comment]
Hi @wongfei,

You can refer to #2355 due to different FOV between Depth and Color.

closed 2018-11-09 by RealSense-Customer-Engineering