realsenseai/librealsense #2238: rs2_pipeline_start_with_config Flush failed-2147024872

wongfei, 2018-08-14:
the case is:

1. app starts with default settings
[RS2] oni::driver::Rs2Device::startPipeline 
[RS2] rs2_create_config
[RS2] enable stream type=3 id=0 640x480 @30
[RS2] enable stream type=2 id=0 640x480 @30
[RS2] enable stream type=1 id=1 640x480 @30
[RS2] rs2_create_pipeline
[RS2] rs2_pipeline_start_with_config
[RS2] STARTED

2. used changes depth video mode, reconfigure pipeline
[RS2] oni::driver::Rs2Stream::stop sensorType=3 streamId=0
[RS2] oni::driver::Rs2Device::updateConfiguration 
[RS2] oni::driver::Rs2Stream::setProperty propertyId=3 dataSize=16
[RS2] oni::driver::Rs2Stream::start sensorType=3 streamId=0
[RS2] oni::driver::Rs2Device::updateConfiguration 
[RS2] oni::driver::Rs2Device::restartPipeline 
[RS2] oni::driver::Rs2Device::stopPipeline 
[RS2] rs2_pipeline_stop
[RS2] rs2_delete_pipeline_profile
[RS2] rs2_delete_config
[RS2] rs2_delete_pipeline
[RS2] STOPPED

3. flush failed error when restarting pipeline with new config
[RS2] oni::driver::Rs2Device::startPipeline 
[RS2] rs2_create_config
[RS2] enable stream type=3 id=0 640x480 @60
[RS2] enable stream type=2 id=0 640x480 @30
[RS2] enable stream type=1 id=1 640x480 @30
[RS2] rs2_create_pipeline
[RS2] rs2_pipeline_start_with_config
[RS2] ERROR at FILE Rs2Device.cpp LINE 288 FUNC oni::driver::Rs2Device::startPipeline
	rs2_pipeline_start_with_config failed: Flush failed-2147024872

only reproduced when changing depth mode

```
OniStatus Rs2Device::startPipeline()
{
	rsTraceFunc("");

	for (;;)
	{
		if (m_pipeline)
		{
			rsTraceError("Already started");
			break;
		}

		Rs2Error e;

		rsLogDebug("rs2_create_config");
		m_config = rs2_create_config(&e);
		if (!e.success())
		{
			rsTraceError("rs2_create_config failed: %s", e.get_message());
			break;
		}

		rs2_config_disable_all_streams(m_config, &e);

		{
			Rs2ScopedMutex lock(m_streamsMx);

			for (auto iter = m_createdStreams.begin(); iter != m_createdStreams.end(); ++iter)
			{
				Rs2Stream* stream = *iter;
				if (stream->isEnabled())
				{
					const OniVideoMode& mode = stream->getVideoMode();

					rsLogDebug("enable stream type=%d id=%d %dx%d @%d", 
						(int)stream->getSensorType(), stream->getStreamId(), mode.resolutionX, mode.resolutionY, mode.fps);

					rs2_config_enable_stream(
						m_config, convertStreamType(stream->getSensorType()), stream->getStreamId(), 
						mode.resolutionX, mode.resolutionY, convertPixelFormat(mode.pixelFormat), mode.fps, &e
					);
					if (!e.success())
					{
						rsTraceError("rs2_config_enable_stream failed: %s", e.get_message());
					}
				}
			}
		}

		// pipeline

		rs2_context* context = getDriver()->getRsContext();
		rsLogDebug("rs2_create_pipeline");
		m_pipeline = rs2_create_pipeline(context, &e);
		if (!e.success())
		{
			rsTraceError("rs2_create_pipeline failed: %s", e.get_message());
			break;
		}

		rsLogDebug("rs2_pipeline_start_with_config");
		m_pipelineProfile = rs2_pipeline_start_with_config(m_pipeline, m_config, &e);
		if (!e.success())
		{
			rsTraceError("rs2_pipeline_start_with_config failed: %s", e.get_message());
			break;
		}

		{
			Rs2ScopedMutex lock(m_streamsMx);

			for (auto iter = m_createdStreams.begin(); iter != m_createdStreams.end(); ++iter)
			{
				Rs2Stream* stream = *iter;
				if (stream->isEnabled())
				{
					stream->onPipelineStarted();
				}
			}
		}

		// depth to color aligner

		m_alignQueue = rs2_create_frame_queue(1, &e);
		if (!e.success())
		{
			rsTraceError("rs2_create_frame_queue failed: %s", e.get_message());
			break;
		}

		m_alignProcessor = rs2_create_align(RS2_STREAM_COLOR, &e);
		if (!e.success())
		{
			rsTraceError("rs2_create_align failed: %s", e.get_message());
			break;
		}

		rs2_start_processing_queue(m_alignProcessor, m_alignQueue, &e);
		if (!e.success())
		{
			rsTraceError("rs2_start_processing_queue failed: %s", e.get_message());
			break;
		}

		rsLogDebug("STARTED");
		return ONI_STATUS_OK;
	}

	stopPipeline();
	return ONI_STATUS_ERROR;
}

void Rs2Device::stopPipeline()
{
	if (m_pipeline) { rsTraceFunc(""); }

	if (m_pipeline)
	{
		Rs2Error e;
		rsLogDebug("rs2_pipeline_stop");
		rs2_pipeline_stop(m_pipeline, &e);
	}
	if (m_alignProcessor)
	{
		rs2_delete_processing_block(m_alignProcessor);
		m_alignProcessor = nullptr;
	}
	if (m_alignQueue)
	{
		rs2_delete_frame_queue(m_alignQueue);
		m_alignQueue = nullptr;
	}
	if (m_pipelineProfile)
	{
		rsLogDebug("rs2_delete_pipeline_profile");
		rs2_delete_pipeline_profile(m_pipelineProfile);
		m_pipelineProfile = nullptr;
	}
	if (m_config)
	{
		rsLogDebug("rs2_delete_config");
		rs2_delete_config(m_config);
		m_config = nullptr;
	}
	if (m_pipeline)
	{
		rsLogDebug("rs2_delete_pipeline");
		rs2_delete_pipeline(m_pipeline);
		m_pipeline = nullptr;
		rsLogDebug("STOPPED");
	}
}
```

RealSense-Customer-Engineering, 2018-08-24:
[Realsense Customer Engineering Team Comment]
@wongfei
from the stream you enabled, it looked weird. Since for the stream index, only the IR has two stream index, others should with index 0. And your enable for Depth type (type=1) with index 1 and other types with index 0, I don't think it should work even for the first round you start the pipeline. The right configuration for three streams on D415 should be as below (I use C++, but you can easily map it to C) :

cfg.enable_stream(RS2_STREAM_DEPTH, 0, 640, 480, RS2_FORMAT_Z16, 30);
 cfg.enable_stream(RS2_STREAM_COLOR, 0, 640, 480, RS2_FORMAT_RGB8, 30);
 cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);

Can you try with the configuration I provided in your code and see if it still doesn't work.

wongfei, 2018-08-24:
I can reproduce it with code:
```
#include <iostream>
#include <librealsense2/rs.hpp>

using namespace rs2;

void test_loop(pipeline& p)
{
	for (int i = 0; i < 10; ++i)
	{
		frameset frames = p.wait_for_frames();
		depth_frame depth = frames.get_depth_frame();
		if (!depth) continue;
		float width = depth.get_width();
		float height = depth.get_height();
		float dist_to_center = depth.get_distance(width / 2, height / 2);
		std::cout << dist_to_center << std::endl;
	}
}

void main()
{
	try
	{
		config cfg;
		pipeline p;

		cfg.enable_stream(RS2_STREAM_DEPTH, 0, 640, 480, RS2_FORMAT_Z16, 30);
		cfg.enable_stream(RS2_STREAM_COLOR, 0, 640, 480, RS2_FORMAT_RGB8, 30);
		cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);

		std::cout << "start1" << std::endl;
		p.start(cfg);
		test_loop(p);

		std::cout << "stop1" << std::endl;
		p.stop();
		cfg.disable_all_streams();

		cfg.enable_stream(RS2_STREAM_DEPTH, 0, 640, 480, RS2_FORMAT_Z16, 60);
		cfg.enable_stream(RS2_STREAM_COLOR, 0, 640, 480, RS2_FORMAT_RGB8, 30);
		cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 30);

		std::cout << "start2" << std::endl;
		p.start(cfg);
		test_loop(p);

		std::cout << "stop2" << std::endl;
		p.stop();
		cfg.disable_all_streams();
	}
	catch (const rs2::error& ex)
	{
		std::cout << ex.what() << std::endl;
	}

	std::cout << "exit" << std::endl;
	return;
}
```

Output:
```
start1
1.968
1.968
1.975
1.962
1.962
1.968
1.968
1.975
1.962
1.975
stop1
start2
Flush failed-2147024872
```

[image]

RealSense-Customer-Engineering, 2018-08-28:
[Realsense Customer Engineering Team Comment]
@wongfei
The second start you configure the pipeline with Depth 60fps + IR 30fps, it is not allowed in the same module (Depth and IR are the same module). You can try on different combination with RealSense Viewer also test on rs-capture with various configuration combination.

wongfei, 2018-08-29:
Got it, thanks for help.

closed 2018-08-29 by wongfei