realsenseai/librealsense #2240: How to know if camera is already streaming

DvdZmd, 2018-08-14:
| Required Info                          |                                         |
|---------------------------------|--------------------------- |
| Camera Model                        | D415/D435                     | 
| Firmware Version                    | 05.09.14.00                     | 
| Operating System & Version  |   Win 10                          |  
| Platform                                  | PC                                   |
| SDK Version                            |  RealSense SDK 2.0         | 
| Language                                |  C                                    | 

Hello again, I was searching around the pipeline and device classes but can't find a busy flag or function. Now i'm thinking how to do that using the the functions for get frames that I have. But while, can you give some help?

Regards

RealSense-Customer-Engineering, 2018-08-16:
[Realsense Customer Engineering Team Comment]
Hi @DvdZmd,

How about checking if any error there like 'rs-color' example code?

AntoineWefit, 2018-08-16:
I remember that I once tried to start a pipeline on a busy camera and the following snippets (cant' remember which one though, might be the first one) returned some kind of error of "busy state" for the related pipeline. Just add this after your `int main(int argc, char * argv[]) try {}`. It can be found in the [rs-pointcloud example](https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp).
```
catch (const rs2::error & e)
{
	std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
	system("pause");
	return EXIT_FAILURE;
}

catch (const std::exception& e)
{
	std::cerr << e.what() << std::endl;
	system("pause");
	return EXIT_FAILURE;
}
```

RealSense-Customer-Engineering, 2018-08-20:
[Realsense Customer Engineering Team Comment]
Hi @DvdZmd,

Still need any further support for this topic?

DvdZmd, 2018-08-21:
Solved!

AntoineWefit, 2018-08-22:
Before closing this issue, can you specify how did you solve your problem ? I'm looking to improve my own code, so I'm curious about your method. 
Without testing it (yet), the following snippet might be a cleaner way to check the state of a pipeline (from [this document](https://github.com/IntelRealSense/librealsense/blob/5c27a7c2b49ce98926fb2b72304ae6490f89427d/include/librealsense2/hpp/rs_pipeline.hpp)) instead of trying to use a pipeline and waiting for an error if it's busy : 
```
/**
        * Return the active device and streams profiles, used by the pipeline.
        * The pipeline streams profiles are selected during \c start(). The method returns a valid result only when the pipeline is active -
        * between calls to \c start() and \c stop().
        * After \c stop() is called, the pipeline doesn't own the device, thus, the pipeline selected device may change in
        * subsequent activations.
        *
        * \return  The actual pipeline device and streams profile, which was successfully configured to the streaming device on start.
        */
        pipeline_profile get_active_profile() const
        {
            rs2_error* e = nullptr;
            auto p = std::shared_ptr<rs2_pipeline_profile>(
                rs2_pipeline_get_active_profile(_pipeline.get(), &e),
                rs2_delete_pipeline_profile);

            error::handle(e);
            return pipeline_profile(p);
        }
```

emredog, 2018-10-25:
Hello,

Is there a plan to provide a proper API call to see if a camera is already in use?

I'm currently using exception handling to this end, and I'm not really happy with it:
```
def is_streaming(device_serial):
    try:
        c = rs.config()
        c.enable_device(device_serial)
        p = rs.pipeline
        p.start(c)
        p.stop()
        return False
    except Exception:
        return True
```