realsenseai/librealsense #2241: Intel RealSence Viewer crash when add playback source...

joelihn, 2018-08-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   | 05.09.14.00 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC  |
| Viewer Version                           | 2.10.0                       | 
| Language                           | C++                      | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, when i start Intel.RealSense.Viewer.exe , add two D435 device, both open Stereo Module and RGB Camera，run for a long time (more then 1 hour), everything is fine. But when i add one more playback source, the program will crashed in a hour. If only add a playback source ,the result is the same, crashed...
I don't know if there's a BUG here？
Thank you in advance !

RealSense-Customer-Engineering, 2018-08-17:
[Realsense Customer Engineering Team Comment]
hi @joelihn,

you mean if you connect 2 D435 on PC can have long run more than 1 hour and if connect 3 D435 on PC and long run will crash in a hour?
do you see any error message or the viewer app just freeze?
and would like to know what's your FW version of D435 camera?

RealSense-Customer-Engineering, 2018-08-23:
[Realsense Customer Engineering Team Comment]
hi @joelihn,

we would like to reproduce, so please help to check if below reproduce steps are correct.
1. connect 2 D435 to PC. (direct connect to PC, no hub)
2. start stream from both camera via realsense viewer.
3. Add Source --->Load Recorded Sequence, and open a .bag file (do we need to play the file or just open and leave it)
4. keep it running for 1 hour and see if it still alive.
thanks.

RealSense-Customer-Engineering, 2018-09-14:
[Realsense Customer Engineering Team Comment]
Hello @joelihn,

Thank you for pointing out this issue!  We were able to replicate the issue with adding a .bag file while streaming RealSense Viewer and will work to create a fix shortly.
Please stay tuned, will update this issue when fix is implemented.

joelihn, 2018-10-08:
wait in hope.

ev-mp, 2018-11-19:
Hello @joelihn , some fixes regarding playback with rosbag were  merged with #2598 PR, and went public with version [2.16.3](https://github.com/IntelRealSense/librealsense/tree/v2.16.3)
Please check and update

dorodnic, 2018-11-28:
If you encounter any additional problems, please open a new issue

closed 2018-11-28 by dorodnic

joelihn, 2018-11-30:
> went

Ok, thanks. @ev-mp @dorodnic