realsenseai/librealsense #2242: playback keep reading the same frames while writing frames to disk by stbi_write_png

hflsdupont, 2018-08-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 5.9.14 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.14.1                        | 

### Issue Description

Hi, I'm currently using playback to read from two rosbag files and synchronize those frames. As far as I don't write those frames to disk by stb, pipe.wait_for_frames() return each frames in .bag file correctly. However, when I try to call stbi_write_png, the next call of playback.resume() and pipe.wait_for_frames() return exactly the same frameset as the previous one. 

```
  auto device = pipe.get_active_profile().get_device();
  rs2::playback playback = device.as<rs2::playback>();

  playback.resume();
  frameset = pipe.wait_for_frames(); 
  playback.pause();
```
'''
void saveImages(video_frame & f, std::stringstream & filename, std::string & dir, std::string & subdir)
{
  std::stringstream path; 
  path << dir << "\\" << subdir << "\\" << filename.str(); 
  
  stbi_write_png(path.str().c_str(), f.get_width(), f.get_height(),
            f.get_bytes_per_pixel(), f.get_data(), f.get_stride_in_bytes());
  
  std::cout << "Saved " << path.str() << std::endl;
}
```

Thanks for the help

hflsdupont, 2018-08-15:
Hi @dorodnic 
Thank you for the quick reply. I noticed ```playback.set_real_time()``` has similar functions as calling ```pause``` and ```resume```, but it doesn't work for me as well. I'll try decreasing playback speed.

dorodnic, 2018-10-11:
Hi @hflsdupont 
Sorry for long delay. We believe this should be now resolved (since 2.16.1 release).
Do you need additional support on this issue?

closed 2018-10-23 by dorodnic