realsenseai/librealsense #2247: Question about multicamera HW setup

wormyrocks, 2018-08-15:
Hi,
I'm curious about how others are setting up sync cables with multicamera arrangements. I currently have 4 Realsense D415 modules and I purchased all the necessary connectors and cables as stated in the multicamera white paper, but I don't see an obvious way to connect multiple grounds and SYNC lines together other than blobbing solder to join 4 crimp ends in a star configuration. That's probably fine, but it's ugly; how are other people pulling it off? I also don't see an obvious way to reliably connect the passive RC filters close to the connectors; it seems like building a PCB and putting the filtering there defeats the purpose, and I don't really want to try and strip away the shielding on the wires to solder on components in parallel.

thanks so much if anyone has a suggestion!

ankarako, 2018-08-20:
Hi @wormyrocks ,

In my point of view, if someone wants to be neat and clean, he/she will have to make PCBs for the filters and the connectors, to achieve a good result, but everything should work with the "messy" blobing also, at least for experimenting.

RealSense-Customer-Engineering, 2018-08-21:
[Realsense Customer Engineering Team Comment]
Hi @wormyrocks,

Do you still have trouble to create and verify the sync cable between multiple cameras?

wormyrocks, 2018-08-21:
Hi,

Nope, I just soldered all the connectors together, seems fine for now.

Thanks!

RealSense-Customer-Engineering, 2018-08-25:
[Realsense Customer Engineering Team Comment]
Sound good, @wormyrocks.
I will close this issue. If you still have any further question, you can still create another ticket.

closed 2018-08-25 by RealSense-Customer-Engineering